LORD-MicroStrain/MSCL #69: Event Trigger Example

woodman120, 2018-12-28:
I need to configure nodes to have event triggering when any accel value reaches some user defined threshold value. Can you point me to or send me some example of how to do that with the MSCL?
Thanks.

woodman120, 2018-12-30:
I have been doing the applyconfig in the past. I don't have the option of using SensorConnect as the end user will not have installed that software, but will be using my app. I am writing in C# but should be able to work this out. Is there a channel mapping that shows how they are assigned? Are they all the same mappings between different node types?

rwslord, 2018-12-30:
Gotcha, sounds like applying the config programatically is the right call for your application.

To get the channels details for a Wireless Node, you can use the [node.features()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessNode.h:WirelessNode.features) function to get a NodeFeatures object. The NodeFeatures object can tell you all sorts of information on what the device supports, including [channels()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Features/NodeFeatures.h:NodeFeatures.channels). This returns a vector of [WirelessChannel](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessChannel.h:WirelessChannel) objects, which have functions such as [name()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessChannel.h:WirelessChannel.name), [id()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessChannel.h:WirelessChannel.id), [channelNumber()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessChannel.h:WirelessChannel.channelNumber), [type()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessChannel.h:WirelessChannel.type), and [description()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessChannel.h:WirelessChannel.description).

woodman120, 2019-01-02:
I just realized that my problem is that when I am using transmitOnly and I have multiple nodes, then my sampling rate maximum needs to be adjusted down based on available bandwidth. The network wouldn't go OK because it couldn't achieve the rate. I seem to remember seeing some documentation about approximate calcs for bandwidth. Do you have an algorithm you recommend for that?

rwslord, 2019-01-02:
The bandwidth is affected by a lot of things, such as the number of enabled channels, the data format, sample rate, burst vs continuous, etc.

Specifically, with event driven sampling, it's even a little trickier, as we don't _really_ know how often the Node will be sampling. In SensorConnect, when event driven is enabled, we show it taking up the full bandwidth, but when more Nodes are added, we drop that down. We are thinking of ways of improving this in the future.

For now, I'd say your best bet is to play with these parameters, and check the network's percent bandwidth and status that it reports back.

If you want to dive in to it all, you can check out MSCL's source code. Specifically the [SyncSamplingNetwork::calculateNetworkValues](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL/source/mscl/MicroStrain/Wireless/SyncSamplingNetwork.cpp#L415) which uses some functionality that is in [SyncSamplingFormulas.cpp](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL/source/mscl/MicroStrain/Wireless/SyncSamplingFormulas.cpp)

woodman120, 2019-01-03:
Trying to test the event triggered mode and getting the following error:
Failed to write the Event Value (EEPROM 340) (EEPROM 356) to Node 61489

Here is the basic code:

            mscl.EventTriggerOptions triggerOpts = new mscl.EventTriggerOptions();
            var trig = new mscl.Trigger(1, WirelessTypes.EventTriggerType.eventTrigger_ceiling, ThresholdG);
            triggerOpts.trigger(0, trig); 
            trig = new mscl.Trigger(2, WirelessTypes.EventTriggerType.eventTrigger_ceiling, ThresholdG);
            triggerOpts.trigger(1, trig);
            trig = new mscl.Trigger(3, WirelessTypes.EventTriggerType.eventTrigger_ceiling, ThresholdG);
            triggerOpts.trigger(2, trig);
            triggerOpts.enableTrigger(0, ShowX && ThresholdTriggered);
            triggerOpts.enableTrigger(1, ShowY && ThresholdTriggered);
            triggerOpts.enableTrigger(2, ShowZ && ThresholdTriggered);
            triggerOpts.preDuration(5000);     
            triggerOpts.postDuration(5000);     
                node.config.inactivityTimeout(7200);
                node.config.dataFormat(mscl.WirelessTypes.DataFormat.dataFormat_cal_float);
                node.config.dataMode(WirelessTypes.DataMode.dataMode_raw);
             node.config.unlimitedDuration(true);
                node.config.dataCollectionMethod(dataCollMethod);
                    node.config.eventTriggerOptions(trigOpts);
                    node.config.samplingMode(WirelessTypes.SamplingMode.samplingMode_syncEvent);
                ChannelMask activeChannels = new ChannelMask();
                activeChannels.enable(1, ShowX);
                activeChannels.enable(2, ShowY);
                activeChannels.enable(3, ShowZ);
                if (ShowX || ShowY || ShowZ)
                {
                    node.config.sampleRate(eSampleRate);
                    node.config.activeChannels(activeChannels);
                }
                node.wirelessNode.applyConfig(node.config);

Any idea what could be causing this?

rwslord, 2019-01-03:
It sounds like you are getting an Error_NodeCommunication exception. Since you are dealing with wireless (2.4Ghz), this can happen for various reasons (range, interference, etc.).

I would imagine if you ran it again, you wouldn't get the same error consistently. There are a couple options here. You can add some retries yourself by catching the exception and trying to apply the configuration again. Or you can use MSCL's built-in retries. It's disabled by default, but you can enable this by setting WirelessNode.readWriteRetries(). See the [FAQ ](https://github.com/LORD-MicroStrain/MSCL/blob/master/FAQs.md#my-wireless-node-wont-respond-to-commands)for an example of this. Keep in mind that if you use MSCL's built-in retries, the applyConfig will potentially block for a longer amount of time, which can make your program look like it's hung if you aren't doing this in another thread.

woodman120, 2019-01-27:
Thanks for the help, I'm starting a new issue that is not related to this one.

closed 2019-01-27 by woodman120