LORD-MicroStrain/MSCL #70: Search for Datalogging Nodes

woodman120, 2019-01-27:
I have been using code that does a getData from the base station and then looks at the data sweeps to see what nodes are out there and get their state. In the past, I only dealt with one node. Now I need to find however many nodes are logging. I find the first node but then it doesn't seem to find any of the others. Can you provide some sample code for how to do this? Here is basically what I have.

                    while(dTime.TotalSeconds < 60 && datalogNodes.Count < dlognodes.Count)
//                    for (int i = 0; i < 20; i++)
                    {
                        mscl.DataSweeps sweeps = baseStation.BaseStation.getData(500);
                        if (sweeps.Count > 0)
                        {
                            foreach (mscl.DataSweep sweep in sweeps)
                            {
                                uint _nodeaddress = sweep.nodeAddress();
                                if (dlognodes.Count(n => n == _nodeaddress) > 0 && datalogNodes.Count(n => n.address == _nodeaddress) == 0)
                                {
                                    foreach (var wdp in sweep.data())
                                    {
                                        if (wdp.channelId() == WirelessChannel.ChannelId.channel_diag_state)
                                        {
                                            DataLogNode node = new DataLogNode();
                                            node.state = wdp.as_uint16();
                                            node.address = _nodeaddress;
                                            datalogNodes.Add(node);
                                            ConsoleMessage += string.Format(_languageMapping.GetTextValue("Console8"), node.address);
                                            retval = true;
                                            starttime = DateTime.Now;
                                            break;
                                        }
                                    }
                                }
                            }
                        }
                        System.Threading.Thread.Sleep(250);
                        dTime = DateTime.Now - starttime;
                    }

woodman120, 2019-01-27:
It works better at lower frequencies so it's just the bandwidth that is slowing it down.

closed 2019-01-27 by woodman120

rwslord, 2019-01-28:
I see you closed the issue, but I'm not sure I completely understood the problem you were having. If you still need help, please let me know!