LORD-MicroStrain/MSCL #71: Setting Acceleration on Nodes

woodman120, 2019-02-02:
We have G-Line-200 sensor nodes with 3 accelerometers in orthogonal directions. I am being asked to write the acceleration value to the node and set it to +-40g. This is part of the configuration settings in Sensor Connect. I don't find an equivalent in the mscl library. I would expect it to be in the node configuration. Is there a way to do this? or does the node already come set for 40g and there is no need to set it?

rwslord, 2019-02-04:
The G-Link-200 ships with a range of +/- 10G. The setting in MSCL that you are looking for is the [`WirelessNodeConfig.inputRange()`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Configuration/WirelessNodeConfig.h:WirelessNodeConfig.inputRange) function.

The code to do this would look something like:
```py

config = mscl.WirelessNodeConfig()

#The input range option requires a channel mask of the channels that are affected by it. 
#For the G-Link-200, this is channels 1,2, and 3.
#The ChannelMask object accepts a decimal value that represents the channels in bits, so using a value of 7 here (0000 0111).
ChannelMask accelChs = new ChannelMask(7);
config.inputRange(accelChs, mscl.WirelessTypes.range_40G);

node.applyConfig(config)

```

woodman120, 2019-02-04:
> For the 'Failed to get the datalogging session info' issue, I'd like to check in on a few things.
>
> Are you able to get to the Download Datalog Sessions screen in SensorConnect?
>
> Is the Node in Idle mode when you are attempting to get the number of datalog sessions? Or is it sampling/datalogging?

The node is in idle and will do transmit only sampling without issue. It also does log only and acts like it is working but I can’t download or get datalog sessions.

rwslord, 2019-02-04:
Ok, can you check if you can get to the Download Datalog Sessions screen in SensorConnect? 

If you can, could you send me your MSCL code, and let me know the line that the issue occurs on? Feel free to email me directly at richard_stoneback@lord.com if you don't want to publish it on here.

woodman120, 2019-02-04:
Using SensorConnect I get the same message.
The message is: Datalog Download - The operation failed: Error: Failed to get the Datalogging Session Info.
It's got to be hardware related, don't you think?

rwslord, 2019-02-04:
hmm, it definitely seems so. Can you send an email to andrew_deering@lord.com, or call us at 802-862-6629, and reference this issue? He will be able to help you out with this. I'll fill him in in the meantime.

closed 2019-02-09 by woodman120