LORD-MicroStrain/MSCL #72: Using DPInst on 32 Bit Machine to Install WSDA200 Base Station

woodman120, 2019-02-09:


woodman120, 2019-02-09:
I have instructions to install the drivers and you sent us the following command line:
DPinst.exe /sw /se /sa /path “<DIRECTORY_TO_WSDA200_DRIVERS>”

So, I have to deal with both 32 bit and 64 bit os while installing the drivers in my setup.exe program.
I created the following vbscript to run:

Set fso = CreateObject("Scripting.FileSystemObject")
Set wshShell = CreateObject( "WScript.Shell" )

If fso.FolderExists(wshShell.ExpandEnvironmentStrings("%windir%") & "\sysnative" ) Then
    wshShell.Run("DPInst\x32\DPInst.exe /sw /se /sa /path ""WSDA200""")
Else
    wshShell.Run("DPInst\x64\DPInst.exe /sw /se /sa /path ""WSDA200""")
End if 

I can run the command for the x32 version on my 32 bit laptop and it runs without any erros. The wsda200 base does not install successfully when I insert the usb base station in the laptop. It gives errors that the driver was not found.

I need to automate this installation. I'm using InstallShield LE which allows me to insert custom actions as vbscripts. That's what I'm doing.

The WSDA200 directory is the upper level directory for the drivers and it has the lordvcp.cat and lordvcp.inf files in it with folders for x64 and x86.

Any ideas what I could be doing wrong here? Any better suggestions?

rwslord, 2019-02-11:
For debugging purposes, can you take out the `/sw` flag from the DPInst command, and see if the dialog wizard that pops up give you any errors?

woodman120, 2019-02-17:
I tried that and it just comes up with a command prompt again. I have also gone into device manager and done a search for drivers and pointed to that directory but it doesn't find drivers for the device. If I could get it to work by command line, then I can make it happen during installation, but I'm striking out.

rwslord, 2019-02-18:
When you remove the `/sw` flag, you should definitely be seeing a DPinst window dialog pop up, and ask you to step through it. Are you seeing this window at all?

If not, can you send me a screenshot of the folder directory, and the command in the command prompt window that you are running?

woodman120, 2019-02-23:
I ended up finding that my scripting would work if I ran as administrator. To make it easier, I just wrote a quick console app and added that to my install. Thanks for the help.

closed 2019-02-23 by woodman120