LORD-MicroStrain/MSCL #73: DLL fails load of _mscl.pyd on Windows 10 and Anaconda3

craiganderson-53546, 2019-02-14:
I have a WSDA-2000, SensorConnect, Anaconda3.  I successfully read data from the WSDA a month ago without issue using Python in Jupyter Notebook.  Then I got a new laptop.  I reloaded the latest SensorConnect and Anaconda3, and did a conda update.  All my other notebooks run fine but my notebook that imports mscl fails.  I downloaded the latest mcsl.py (3.0.6) into ..\Anaconda\lib and _mscl.pyd into ..\Anaconda3\DLLs, both paths appear in sys.path.  The error occurs in mscl.py as it tries to load _mscl.pyd.  The same error occurs from a py script run from the Anaconda prompt.  The error appears as follows...

ImportError                               Traceback (most recent call last)
~\AppData\Local\Continuum\anaconda3\lib\mscl.py in swig_import_helper()
     17         try:
---> 18             fp, pathname, description = imp.find_module('_mscl', [dirname(__file__)])
     19         except ImportError:

~\AppData\Local\Continuum\anaconda3\lib\imp.py in find_module(name, path)
    295     else:
--> 296         raise ImportError(_ERR_MSG.format(name), name=name)
    297 

ImportError: No module named '_mscl'

During handling of the above exception, another exception occurred:

ImportError                               Traceback (most recent call last)
<ipython-input-3-3b6a8703300a> in <module>
----> 1 import mscl

~\AppData\Local\Continuum\anaconda3\lib\mscl.py in <module>
     26                 fp.close()
     27             return _mod
---> 28     _mscl = swig_import_helper()
     29     del swig_import_helper
     30 else:

~\AppData\Local\Continuum\anaconda3\lib\mscl.py in swig_import_helper()
     18             fp, pathname, description = imp.find_module('_mscl', [dirname(__file__)])
     19         except ImportError:
---> 20             import _mscl
     21             return _mscl
     22         if fp is not None:

ImportError: DLL load failed: The specified module could not be found.

rwslord, 2019-02-14:
I don't have any experience with Anaconda, but in general, I always have `_mscl.pyd` sitting directly next to `mscl.py`. Could you try copying `_mscl.pyd` into the `..\Anaconda\lib` folder as well, and see if that helps?

craiganderson-53546, 2019-02-15:
Already tried that, no love.

Sent from my iPhone, please excuse spelling and brevity

rwslord, 2019-02-15:
Can you see if you can import mscl from a python file that is directly next to mscl.py and _mscl.pyd?

You also mentioned a version 3.0.6. Is that your python version? MSCL is up to 48.0.0, and it's compiled with Python 2.7, and 3.6. I'm not sure if mscl.py for Python 3.6 will work with python 3.0.6.

craiganderson-53546, 2019-02-15:
Yes, just verified with both files in the same location and the symptom persists.  The version string I gave was in the mscl.py file header, perhaps not useful.  I am running Python 3.7.1

rwslord, 2019-02-15:
I've replicated your issue. I'm looking into it now.

rwslord, 2019-02-15:
So it looks like, since MSCL was built with Python 3.6, it can't be imported using 3.7, unfortunately. I would have thought this would work, but there is specifically a Python36.dll that it needs.

So unfortunately, I think the only options here are to use Python 3.6, or to [compile it yourself](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Windows.md#building-mscl-for-python) using Python3.7.

craiganderson-53546, 2019-02-15:
Yeah, makes sense.  I tried with the release of mscl that was current when I originally started this project, then I looked for a history of versions of Python to see what I had installed before. I will try to recompile it first.  Thanks for the help!

rwslord, 2019-02-15:
no problem!

closed 2019-02-15 by rwslord

craiganderson-53546, 2019-02-15:
AH!  Way easier to simply create a 3.6 env within my existing Anaconda (e.g., conda create <<new_name>> python=3.6).  Instructions are available [here](https://conda.io/projects/conda/en/latest/user-guide/getting-started.html#managing-python)