LORD-MicroStrain/MSCL #74: Memory of the G-Link 200

pa2492, 2019-02-25:
Good afternoon , 
We are using the MSCL LABVIEW example that is posted MSCL Acquire data.VI  to read data from G-link 200  sensors. However, after 5 mins of running we notice high CPU usage in Labview, as if the data that is being transmitted to the WSDA is quite big since at every iteration , labview is reading the same data + a small new dataset.....Altough I use the "overwrite" option as the storage mode , is there any way we can wipe the data from the sensor once transmitted? And how many samples does the memory of the G -link 200 nodes supports before it start wiping ? 
Thanks

rwslord, 2019-02-25:
The data being transmitted from a Wireless Node to a WSDA will never be replicated as you are suggesting (sending the old data, plus new data). So there must be something else going on here.

In addition, just to clarify some things, the "overwrite" storage mode option only affects log mode, which I don't believe is what you are doing. You are using transmit mode, so the data is not being logged to the Node, but being transmitted in real-time over the air.

We will attempt to replicate your issue, and get back to you. In the meantime, can you please send a screenshot of the "processes" tab in your task manager that shows the process name, and its CPU usage.

pa2492, 2019-02-25:
Thanks Robert for reaching out. Here  is the screenshot.
Thanks again
Patrick

[image]

pa2492, 2019-02-25:
Not sure if this helps for replication, attached are how I am configuring the sensor. And again I am using the MSCL Example Acquire Data.vi  example. 
Also, what I meant by reading old data  + new ...I meant reading from hte base station and not the sensor node.... GET DATA SWEEP function in the below figure. This function after each iteration might contain old data. (  correct me plz if I am wrong) and I appologise for the confusion . 
Thanks a lot 

[image]
[image]
[image]
[image]

rwslord, 2019-02-25:
Thanks for the CPU screenshot. Could you send another that shows the "Processes" tab active, sorted by CPU, so that I can see the processes that are using the top most CPU?

Successively calling the getData function on the BaseStation object will not return duplicate data. Once you call getData and retrieve some data points, they are removed from MSCL's internal buffer.

pa2492, 2019-02-25:
Hi Robert, 
Attached is the processes image. Thanks for the explanation, the reason why I thought when calling getData function the data from previous iteration are still there is because they are kept shown in the XY graph function .
Thanks again for your help and prompt assistance. 
Columbia monitoring team 

[image]

rwslord, 2019-02-27:
I'm sure our customer support team will reach out to you as well, since this conversation was continued there. It turns out the example code was continuously appending data to a buffer, which was causing things to eventually slow down. A new version has been released which should resolve this issue: https://github.com/LORD-MicroStrain/LabVIEW-MSCL/releases/tag/v0.6.1.4

closed 2019-02-27 by rwslord

pa2492, 2019-02-27:
Thanks a lot Richard for following up, I'll take a look on it.

pa2492, 2019-03-05:
Hi Robert, 
Thanks for the new version of MSCL example. The previous examples before updating as you said they were continuouslt appending data to a buffer which was causing things to slow down. However, the new version do not append at all. For the sake of our project, we would like to append a data to a buffer but not continuously and forever as the old version did... Maybe store 10 reading requests and then wipe the buffer ... Can you advise what should I change in the code example to customize a buffer? Is this doable in the new version? 
Big thanks

rwslord, 2019-03-05:
However much data you decide to store, and what you would like to do with it, is completely up to you. The LabVIEW mscl code is just meant as an example for you to take and expand upon as needed. Unfortunately, we can't provide a lot of support in development of your own software, but you should be able to take the data retrieved from MSCL and put them in whatever types of LabVIEW containers suit you.

pa2492, 2019-03-05:
True thanks