LORD-MicroStrain/MSCL #75: Catch in Labview when connection with G-link node is lost

pa2492, 2019-03-05:
Dear software team, 
In the labview example posted  MSCL Acquire data.vi I am trying to edit it to capture with a boolean when the WSDA loose connection with the G-Link as shown in the figure below. However, the node address returns always zero. 
Any idea how I can keep track in my code of when one of the nodes is disconnected? 
Thanks 
[image]

rwslord, 2019-03-05:
There is no real connection between a Node and WSDA. The gateway is simply just listening to the Node transmit data. Can you explain what you are trying to look for?

pa2492, 2019-03-05:
Thanks Robert for your prompt reply. Basically if these sensors are on site monitoring equipements 24/7 , and the gateway is getting data from, we would like to know when we loose a sensor maybe from battery or any other reason. In other words, we would like to check continuously if this sensor is still sampling and sending data.

rwslord, 2019-03-05:
The only way I can think to do this is to keep a list of active nodes, and how often you expect to hear data from them (sample rate). Then, on a schedule, you would need to check the last time you heard data from each sensor, and send an alert or do whatever action you want, when a Node hasn't been heard for a certain period of time. This isn't something built into MSCL or our LabVIEW example code, so you will have to develop it yourself.

mglord, 2021-06-15:
The above-mentioned strategy (keep track of which nodes you've received data from recently) is the most straight-forward way to tell if the base station has lost communication with a node for any reason.

Additionally, most of our newer nodes report battery status within the diagnostic information. Nodes can be configured to output diagnostic data packets at a regular interval so if battery power is the primary concern you can detect if a node is reporting low or critical battery on supported devices.

Hope this helps! I'm going to close this but feel free to re-open if you continue to run into issues or anything needs clarification. Thanks!

closed 2021-06-15 by mglord