LORD-MicroStrain/MSCL #76: Trouble building on NI Linux Real-Time

jrmlgt, 2019-03-07:
I've downloaded the manual build script and modified it to use okpg instead of apt or yum (see [MSCL_Install.txt](https://github.com/LORD-MicroStrain/MSCL/files/2941362/MSCL_Install.txt)).
When trying to call bjam in the buildMSCL function, it tells me it doesn't find boost-build.jam
```
Unable to load Boost.Build: could not find "boost-build.jam"
---------------------------------------------------------------
Attempted search from /usr/share/mscl/programs/MSCL-BUILT up to the root
at /usr/share/mscl/programs/share/boost-build
and in these directories from BOOST_BUILD_PATH and BOOST_ROOT: /usr/share/boost-build.
Please consult the documentation at 'http://www.boost.org'.
Unable to load Boost.Build: could not find "boost-build.jam"
---------------------------------------------------------------
```
I've then modified the command
`mscl_call = 'MSCL//stage_python release ' + flags + ' -j2'`
to the following to manually add the path.
`mscl_call = 'MSCL//stage_python release ' + flags + ' -j2 -sBOOST_BUILD_PATH=' + glob(os.path.join('/usr/share/' + configDict['package_name'] + '/programs/', 'boost*'))[0]`

It responds me with the following error:
```
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/tools/gcc.jam:164: in gcc.init from module gcc
error: toolset gcc initialization:
error: provided command 'g++' not found
error: initialized from ../project-config.jam:1
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/build/toolset.jam:44: in toolset.using from module toolset
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/build/project.jam:1102: in using from module project-rules
../project-config.jam:1: in modules.load from module project-config
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/build-system.jam:255: in load-config from module build-system
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/build-system.jam:486: in load-configuration-files from module build-system
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/build-system.jam:607: in load from module build-system
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/kernel/modules.jam:295: in import from module modules
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/kernel/bootstrap.jam:139: in boost-build from module
/usr/share/mscl/programs/boost_1_68_0/boost-build.jam:17: in module scope from module
```

It doesn't seem to find the g++ command.
How can I overcome this?

jrmlgt, 2019-03-07:
I've tried it again and the error appears actually earlier, when using b2 in the init function.
`subprocess.call(['./b2', 'install', '--prefix=/usr/share/' + configDict['package_name'] + '/programs/', '-j2'], cwd=directory)`
```
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/tools/gcc.jam:230: in gcc.init from module gcc
error: toolset gcc initialization:
error: no command provided, default command 'g++' not found
error: initialized from project-config.jam:12
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/build/toolset.jam:44: in toolset.using from module toolset
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/build/project.jam:1102: in using from module project-rules
project-config.jam:12: in modules.load from module project-config
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/build-system.jam:255: in load-config from module build-system
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/build-system.jam:486: in load-configuration-files from module build-system
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/build-system.jam:607: in load from module build-system
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/kernel/modules.jam:295: in import from module modules
/usr/share/mscl/programs/boost_1_68_0/tools/build/src/kernel/bootstrap.jam:139: in boost-build from module
/usr/share/mscl/programs/boost_1_68_0/boost-build.jam:17: in module scope from module
```

rwslord, 2019-03-07:
> CAUTION: This email originated from a sender outside LORD Corporation. Do not click links or open attachments unless you recognize the sender and know the content is safe.

The error it is giving you is “g++ not found”. You should google this error code for your specific version of linux you are compiling for, and install whichever package is recommended.

ardLORD, 2019-03-07:
The error it is giving you is “g++ not found”. You should google this error code for your specific version of linux you are compiling for, and install whichever package is recommended.

Thank you,
Andrew Deering
Technical Support Engineer, Sensing Systems
+1(802)876-6546<tel:18028766546>
Andrew.Deering@Lord.com<mailto:ardeering@microstrain.com>

LORD Corporation
MicroStrain Sensing Systems
459 Hurricane Lane, Suite 102
Williston, VT 05495 USA
www.microstrain.com<http://www.microstrain.com/>
www.lord.com<http://www.lord.com/>

Please click here to rate the support you received!<https://www.surveymonkey.com/r/T2KPV6D>

jrmlgt, 2019-03-08:
There was visibly a trouble with the g++ symlinks which I was able to resolve with
`opkg install --force-reinstall g++-symlinks`
The b2 function now works propertly.
But when I try the bjam function, it responds me with this:
```
swig.swig /usr/share/mscl/programs/MSCL-BUILT/build/MSCL/source/mscl/Wrapper/MSCL_Main_Interface.cpp
gcc.compile.c++ /usr/share/mscl/programs/MSCL-BUILT/build/MSCL/gcc-5.3.0/release/source/mscl/Wrapper/MSCL_Main_Interface.o
virtual memory exhausted: Cannot allocate memory
```
I've seen the same problem on [issue 65](https://github.com/LORD-MicroStrain/MSCL/issues/65).
I've tried to remove the -j2 option but it is not sufficient.
My device has 1 Go of ram.

Do you know of another workaround to decrease the necessary memory?

rwslord, 2019-03-08:
The only things I can think of are:

1. stopping all other services that are running that aren't needed
2. configure a swapfile
3. cross-compile

Unfortunately, I don't have direct experience with these, but I think these are some good options

mglord, 2021-06-15:
I hope you were able to get this working! Just wanted to call out that we now offer [pre-built packages](https://github.com/LORD-MicroStrain/MSCL#pre-built-mscl-binariespackages-v6200) for various linux distros.

I'm going to go ahead and close this, but I feel free to re-open or enter another issue if you continue to run into problems! Thanks!

closed 2021-06-15 by mglord