LORD-MicroStrain/MSCL #77: Disovering Nodes Help

Testing1User, 2019-04-03:
Hi,

I am new to coding and would like some directions on discovering new nodes. I have a WSDA-200-USB and have been trying to search for available nodes.
Compiled in RPI 3.

My code:

`import time
import sys
sys.path.append('/home/pi/Desktop/MSCL')
import mscl
import setup

try:
    connection = mscl.Connection.Serial(setup.COM_PORT, setup.Brate)
    baseStation = mscl.BaseStation(connection)
    while True:
        x = baseStation.getNodeDiscoveries()
        for NodeDiscovery in x:
            print  NodeDiscovery.NodeAddress(),
            time.sleep(1)
except Exception, e:
    print "Error:" , e`

Any pointers to the right direction would be highly appreciated!
Thank you.

rwslord, 2019-04-03:
Does your code run ok, or do you get some errors?

Nodes will only send out Node Discovery packets when they are first powered on. The way that the getNodeDiscoveries() function works in MSCL is that it returns a list of Node Discoveries that were heard since the BaseStation object was created. So, for instance, if you created your mscl.BaseStation object, and then sleep/wait in your program, physically cycled the power on all the Nodes you want to discover, and then proceed with your program, and then call getNodeDiscoveries, you should see all of them in there. 

You can also through this functionality into a separate thread, and just continuously call it, if you wanted to be notified of Node Discoveries at all times.

Testing1User, 2019-08-09:
Hi!

Sorry the project was pushed aside for a while but I am back at it now.
Thanks for the advice it works great. 

I used this code in case anyone else wants it - 
`#!/usr/bin/python
import time
import sys
import mscl
import Setup

try:
    connection = mscl.Connection.Serial(Setup.COM_PORT, Setup.Brate)
    baseStation = mscl.BaseStation(connection)

    while True:
        sweeps = baseStation.getNodeDiscoveries()
        for sweep in sweeps:
            print "Packet Received",
            print "Node:", sweep.nodeAddress()

except Exception, e:
print "Error: ",e
`

closed 2019-08-09 by Testing1User

vinod743374, 2021-11-09:
can you please tell us .. how do u import setup ??