LORD-MicroStrain/MSCL #78: how to use ShuntCalCmdInfo

AndersonNontol, 2019-04-05:
I am writing some code in Python 3.6 to perform an external shunt calibration using MSCL but, I do not understand how to use ShuntCalCmdInfo. I have coppied what is at the bottom of the next link but it does not work. http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/Commands/AutoCalInfo.h:ShuntCalCmdInfo.ShuntCalCmdInfo

Do I have to do any change on the code that appears in the link in order to work with pyhton?

Additional information: I am using V-link-200, and want to perform an external shunt calibration with an external excitation voltage (not 4.096V)

rwslord, 2019-04-05:
I'm currently out of the office, so I can potentially help more on Monday once I have more details, but I'm hoping to get you started in the right direction at least.

The command you are looking for is [`WirelessNode.autoShuntCal()`](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Wireless/WirelessNode.h:WirelessNode.autoShuntCal).

This command takes a ChannelMask object representing the channel to calibrate, as well as a `ShuntCalCmdInfo` object, which is what you sent a link to.

The code to calibrate autocalibrate a channel will look something like this:
```py

#create a channel mask object for channel 1
ch1 = mscl.ChannelMask()
ch1.enable(1)

#create a ShuntCalCmdInfo object and adjust with appropriate values
cmd = mscl.ShuntCalCmdInfo()
cmd.useInternalShunt = False
cmd.numActiveGauges = 0
cmd.gaugeResistance = 0
cmd.shuntResistance = 0
cmd.gaugeFactor = 0.0
cmd.inputRange = mscl.WirelessTypes.range_39mV # type of WirelessTypes.InputRange
cmd.hardwareOffset = 0
cmd.excitationVoltage =  mscl.WirelessTypes.voltage_2800mV #type of WirelessTypes.Voltage

result = node.autoShuntCal(ch1, cmd)

#the autoShuntCal command returns an AutoShuntCalResult object
result.errorFlag()
result.slope()
result.offset()
result.baseMedian()

```

It might help to look at SensorConnect to see the possible options.

Also keep in mind that, depending on what your application is doing, you might be able to get away with just performing the shunt cal within SensorConnect, and then doing everything else in your application (since configuration settings and calibration are saved in between power cycles).

mglord, 2021-06-15:
As mentioned, the best option is often to perform the shunt cal in SensorConnect, but if you need to run it in your application the `autoShuntCal()` function will run the calibration operation on the device and return the slope and offset coefficients to write to the device. Additional information on the function and parameters can be found [here](https://github.com/LORD-MicroStrain/MSCL/issues/162#issuecomment-859777447) - some of this info is specific to the SG-Link-200-OEM but we would be happy to clarify as needed.

I'm going to go ahead and close this. I hope you were able to get this working, but if you have any outstanding questions or issues feel free to re-open this or enter another issue! Thanks!

closed 2021-06-15 by mglord