LORD-MicroStrain/MSCL #79: G-Link Config and Data Download Issues

aibanizu79, 2019-04-19:
Hi, I am having the following issues when configuring, logging and downloading data from the nodes.
I am running the mscl on Windows and Python.
I have two G-Link nodes configured to Sync mode, sample rate 256, data format to Cal Float, transmit is log only and capturing 2100 sweeps, unlimited duration is off.

What I am trying to do is to Start Sampling on demand, and capture X number of samples from the two Synchronised nodes. Then plot this data or save it to .txt or .csv.

I configure the two nodes and the network and start sampling. Then I am using the DataDownloader to get data from the nodes, I have used examples from this forum to build my code. Here are the problems observed:
1. Only one node logs data
2. The data points are not saved as Cal Float
3. This node then captures random number of sessions (sometimes 1 other times 3-4), instead of one
4. The timestamp from the sweep.timestamp is way off from the real time (is this node or base config? or sweep config?

I don't think I am understanding how the DataDownloader, getNextData and startOfSession actually work to extract or save the data. Any help is appreciated.

rwslord, 2019-04-20:
Could you send me your code? You can email me at richard_stoneback@lord.com if it makes it easier.

It also would be helpful for debugging if you could try using SensorConnect and see if you can replicate the same issue there. In SensorConnect, plug in your BaseStation, power on each node 
so you see them in the device list, then click on the BaseStation and choose the Sampling button. You should then be able to see exactly how each Node is configured and confirm it's how you expect.

aibanizu79, 2019-04-20:
Hi Richard, thanks for your prompt reply. I'll email you shortly.
The sensors work fine on SensorConnect. The node and base config, sampling and visualization. I also have a G-link-200-OEM working fine. I'll explain in my email.

closed 2019-06-08 by aibanizu79