LORD-MicroStrain/MSCL #80: MSCL C++ problems with building wireless examples

AlexanderMoran1995, 2019-04-22:
Is it possible to build the MSCL library for c++ with visual studio versions newer than 14.0? Such as 14.1 or 14.2? Also, Can you use it with boost versions other than 1.68? or does the MSCL library only work explicitly with msvc 2015 and boost 1.68? 

I have been trying to get the examples to run with both the prebuilt example solution and by including the files within the example to my own blank project with no success. Ive gotten varying degrees of errors most of which Ive been able to sort out but the most persistent one is telling me that the MSCL library was built using an older version of the compiler than the rest of my libraries. That being said i was wondering could the issue be solved by building the MSCL library from source with a newer version of visual studio like 14.2 or 14.1?

I have tested my build of boost 1.68 individually with some of their examples and they run properly so I have eliminated that as a possible cause of my build errors.
on a side note could you elaborate some on what to do for this step and where to find this file in order to edit it?  

**Edit the MSCL_PropertySheet_Debug.props and MSCL_PropertySheet_Release.props files, or the property pages of each project**

rwslord, 2019-04-22:
You should definitely be able to build MSCL with a newer version of Visual Studio/MSVC. It sounds like that is likely the issue you are running into, but feel free to send the errors if we want me to check them out.

As a note here, if you install the older Visual Studio (2015) toolset on your machine, you can create a new project using the new Visual Studio, but target the older version when building by setting the Platform Toolset in the properties.

Sorry about those instructions...those are outdated. I forgot to update that readme. You'll want to update the **Additional Include Directories** and the **Additional Dependencies** sections in the MSCL_Wireless_Example_C++ project properties. I'll get that fixed in the next update

AlexanderMoran1995, 2019-04-23:
I tried building the example solution with the older version of the toolset like you suggested which got rid of some errors but was still having a few. After a few different attempts to get it working I finally got it to build successfully after i downloaded a different version of boost. the one I had been using only had the mt-gd option for the .lib file. The other version I installed today had the mt-sgd option and using that has made all the difference.

Thank you for your help and timely responses to my questions. If I have future questions that are not necessarily bugs should I still post them here or is there a more appropriate place for me to post those?

rwslord, 2019-04-23:
Great to hear you are up and running!

GitHub is definitely a good spot for questions, even if they aren't bugs, just so other users can reference them as needed. If you prefer to not post publicly, feel free to email me at richard_stoneback@lord.com.

closed 2019-04-23 by rwslord