LORD-MicroStrain/MSCL #81: converting data type of variables within parseData.h

AlexanderMoran1995, 2019-05-24:
Is it possible to convert the data type of the data elements received from a wireless node to a string? I see that some of the data sweep elements have a .str() or a .prettystr() at the end of them. I tried inserting that at the end of some of the other elements but that resulted in the errors in the attached image. Im attempting to insert all of the node data into a vector to be sorted and manipulated later.  Mainly just looking for a convenient way to get incoming data from each node into a dynamically sized data structure.
[image]

rwslord, 2019-05-24:
I would look into the std::to_string() documentation. But also, you might just want to think about having a vector of objects instead, so that you don't have to cast everything to a string. But really, that's what the sweeps object is, so maybe just pass that around to where you actually want to handle the data?

AlexanderMoran1995, 2019-05-28:
alright ill look into that thanks for the insight! I just wanted to get your input before i went off and reinvented the wheel.

closed 2019-05-28 by rwslord