LORD-MicroStrain/MSCL #82: readEeprom

satriton, 2019-05-28:
i try read the slope and ofset of a eeprom (g-link 200 OEM node), but readEeprom() function only work at uint16 format  (wirelessNode:: class), and the format es a float data (memory 152 for slope ch1 and 156 for offset ch1). I am trying to reconstruct a float from 2 uint16 (152 and154 for slope) but the data  is wrong. Exist a function for read eeproim on float format for Python? Iam using MSCL for python/UbuntuServer16LTS,

rwslord, 2019-05-28:
There is no built in function for reading a float value from eeprom. 

Can I see your code for putting the two values together into a float?

satriton, 2019-05-28:
`           nodes = self.wirelessNodes[::]
            tmp = []
            for node in nodes:
                for location in (152, 154, 156, 158):
                    value_eeprom = node.readEeprom(location)
                    tmp.append(value_eeprom)
            tmp = numpy.array(tmp, numpy.uint16)
            tmp.dtype = numpy.float32
            print("Data EEPROM Node %s is %s" % (node.nodeAddress(), tmp))`

satriton, 2019-05-28:
nodes = self.wirelessNodes[::]
            tmp = []
            for node in nodes:
                for location in (152, 154, 156, 158):
                    value_eeprom = node.readEeprom(location)
                    tmp.append(value_eeprom)
            tmp = numpy.array(tmp, numpy.uint16)
            tmp.dtype = numpy.float32
            print("Data EEPROM Node %s is %s" % (node.nodeAddress(), tmp))

rwslord, 2019-05-28:
I would recommend making it easier and using MSCL's build in methods for this, instead of reading eeproms manually. This would look something like this:

```py
#get the LinearEquation object for channel 1
eq = node.getLinearEquation(mscl.ChannelMask(1)) 
slope = eq.slope()
offset = eq.offset()

#get the LinearEquation object for channel 4 (note that the ChannelMask object takes a mask value, not a channel number)
eq = node.getLinearEquation(mscl.ChannelMask(8)) 
slope = eq.slope()
offset = eq.offset()
```

Note: The ``mscl.ChannelMask`` constructor takes a bitmask representing the channels, instead of a channel number. It can be a bit confusing at first. Here are some other ways to create the ChannelMask object:

```py
ch4Mask = mscl.ChannelMask(8)  #same as above (00001000 = 8 = ch4)

ch4Mask = mscl.ChannelMask( int('00001000', 2) ) #use python's conversion of a bitmask string to int

ch4Mask = mscl.ChannelMask() #start with all channels disabled (0)
ch4Mask.enable(4) #ChannelMask's enable() function takes the channel number itself, so this turns on the bit for channel 4
```

satriton, 2019-05-28:
most better!
thanks.

closed 2019-05-28 by satriton