LORD-MicroStrain/MSCL #83: Sampling Rates and Polling Data

tuckduck, 2019-06-07:
I am using a 3DM-GX4-25 to read Euler Angles in Python 2, and would like to be able to choose the sampling rate as an argument the script is run. The first approach I took was to try to change the sampling rate. However, entering a sampling rate other than 500Hz for the Euler Angles channel causes InertialNode.setActiveChannels(...) to throw Sensor Message format failed. Next I also tried to use pollData at my preferred sampling rate to just grab an individual packet at the desired rate. In the documentation it says pollData takes a MipChannelFields object, for which there is no documentation. I have tried simply instantiating an mscl.MipChannelFields() object and appending the appropriate Channel Field enum, however when this is passed into pollData it returns None type. How can I fix either of these issues to achieve a selectable sampling rate?

tuckduck, 2019-06-07:
I have achieved this to some degree by using InertialNode.getDataPackets() at the selected rate, and just .pop() off the last packet, which I assume to be the most recent. But This seems inefficient since many more packets must be sent than necessary.

rwslord, 2019-06-07:
You should definitely be able to have the sample rate be configurable via MSCL. What sample rate are you trying to enter for the Euler Angles? Can I see the code you are trying to use to change the rate?

tuckduck, 2019-06-07:
`imu = mscl.InertialNode(mscl.Connection.Serial('ttyACM1', 115200))

ahrsImuChs = mscl.MipChannels()

ahrsImuChs.append(mscl.MipChannel(mscl.MipTypes.CH_FIELD_SENSOR_EULER_ANGLES, mscl.SampleRate.Hertz(samp_rate)))

imu.setActiveChannelFields(mscl.MipTypes.CLASS_AHRS_IMU, ahrsImuChs)

imu.enableDataStream(mscl.MipTypes.CLASS_AHRS_IMU)`

where samp_rate is a parameter taken in by the calling method. Whenever it is something other than 500 setActiveChannelFields throws Sensor Message Format failed (error code 4). I found 500 by using imu.getDataRateBase(mscl.MipTypes.CLASS_AHRS_IMU)
--

rwslord, 2019-06-07:
ok, that all looks ok. What sample rates have you tried? Does it error out if you pass in 100?

tuckduck, 2019-06-07:
I'll update this comment in an hour or so when I have access to the IMU to try

mglord, 2021-06-15:
As mentioned previously, you should definitely be able to configure the device to output data at a reduced rate. If you continue to run into trouble with this please let us know!

**Re: poll command**
If you need only a single sample at irregular intervals the poll data command may be a good option. Basically the behavior around this command is:

- Send poll command to device (`InertialNode::pollData()`)
- Receive success/failure response from the device
  - this does not contain any data
  - the `pollData()` function will not return until a response is received or the command times out (default 250 ms)
- Device outputs requested data packet in regular data stream, parsed, and added to MSCL buffer
- Get packets from buffer with `InertialNode::getDataPackets()`

The `pollData()` function takes two parameters, the `MipTypes::DataClass` of the requested data, for example: `MipTypes::CLASS_AHRS_IMU`, and a list of the actual requested fields. The following will poll the device for the LLH Position and NED Velocity from the Estimation Filter for devices that support these fields:
```
// where node is an InertialNode object
node.pollData(
    mscl::MipTypes::CLASS_ESTFILTER, // MipTypes::DataClass
    { // list of MipTypes::ChannelField values
        mscl::MipTypes::CH_FIELD_ESTFILTER_ESTIMATED_LLH_POS,
        mscl::MipTypes::CH_FIELD_ESTFILTER_ESTIMATED_NED_VELOCITY
    }
);

// get all collected data (polled and streamed)
mscl::MipDataPackets pkts = node.getDataPackets();
```

I hope you were able to find a solution that worked for you! I'm going to close this, but if you have any outstanding questions or are running into any additional issues feel free to re-open this or enter another issue. Thanks!

closed 2021-06-15 by mglord