LORD-MicroStrain/MSCL #84: Unable to communicate with 3DM-GX3-25

vibhoraggarwal, 2019-07-19:
I am using [3DM-GX3-25](https://www.microstrain.com/inertial/3DM-GX3-25) on Ubuntu 19.04. For this, I manually built all the c++ dependencies using [this information](https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Linux.md)

Now, I am trying to run one of the c++ examples from [here](https://github.com/LORD-MicroStrain/MSCL/tree/master/MSCL_Examples). The port name I am using is `/dev/ttyACM1` which I checked using `lsusb` by connecting and disconnecting the IMU. But still, I am receiving the following error:
`Node Information: 
Model Name: Error: Failed to communicate with the device.
sh: 1: pause: not found`
I tried to search for this error, but couldn't find a solution. Can anybody suggest a solution?

vibhoraggarwal, 2019-07-22:
I checked if the port from the IMU is sending me any data by using following Matlab commands: 
`serial(); fopen(); fclose()`
 It gives me the result below. So, I believe that the port is sending me data, but am not sure why I am not able to access it through the c++ program.

Matlab data feed:
```204
59
178
162
129
59
242
250
106
63
127
197
90
59
110
145
24
186
202
90
183
59
116
93
84
62
164
6
16
191
11
21
111
190
163
100
172
62
251
238
80
191
94
219
228
187
206
49
111
191
94
214
108
190
251
213
247
188
157
15
240
60
94
191
194
60
116
78
160
191
127
242
168
1
76
46
109
40
121
204
59
187
15
102
60
5
194
141
63
128
34
115
60
36
21
95
57
118
56
248
59
255
213
189
62
163
214
2
191
11
44
129
190
163
112
202
62
251
240
164
191
94
219
57
187
206
113
250
191
94
213
174
190
251
216
30
188
157
197
46
60
95
234
48
60
117
30
56
191
127
242
138
1
76
48
222
38
195
204
59
175
82
195
60
5
83
195
63
127
156
223
185
87
149
54
59
254
175
45
58
129
11
120
62
163
237
216
191
11
75
175
190
163
77
248
62
251
251
115
191
94
216
62
187
201
214
194
191
94
210
159
190
251
227
75
188
157
72
109
60
96
45
35
60
114
167
168
191
127
242
172
1
76
51
79
40
166
204
59
154
183
96
59
255
129
110
63
128
2
174
187
38
213
91
60
22
141
250
188
24
111
93
62
163
219
108
191
11
47
25
190
163
86
118
62
252
13
107
191
94
211
46
187
200
92
244
191
94
205
147
190
251
245
117
188
156
197
0
60
95
155
130
60
113
138
229
191
127
242
198
1
76
53
192
39
173
204
59
182
9
232
59
214
184
45
63
127
133
215
186
221
126
138
59
146
130
92
187
131
238
105
62
163
218
164
191
11
31
7
190
163
114
114
62
252
25
147
191
94
207
192
187
199
131
155
191
94
202
40
190
252
1
187
188
156
117
191
60
95
64
110
60
112
228
96
191
127
242
213
1
76
56
49
40
104
204
59
194
170
252
60
1
196
211
63
128
38
129
59
236
93
154
187
6
9
101
187
100
164
63
62
163
169
114
191
11
46
119
190
163
70
120
62
252
41
5
191
94
203
96
187
200
73
135
191
94
197
193
190
252
17
8
188
156
233
78
60
95
208
133
60
113
180
13
191
127
242
192
1
76
58
162
38
1
204
59
210
207
22
59
241
253
121
63
127
76
255
60
26
33
97
59
158
30
76
187
192
74
168
62
163
157
248
191
11
38
227
190
163
79
72
62

rwslord, 2019-07-22:
@vibhoraggarwal - are you using USB or is it a direct serial connection?

vibhoraggarwal, 2019-07-22:
@rwslord I am using USB connection

rwslord, 2019-07-22:
ok, can you try passing in a baud rate to the connection line? So something like this:

```cpp
mscl::Connection connection = mscl::Connection::Serial("/dev/ttyACM1", 115200);
```

vibhoraggarwal, 2019-07-22:
I tried to pass the baud rate the way you mentioned. I passed both 9600 and 115200 as baud rate (because, when I try to read the serial data in Matlab, it shows the baud rate as 9600). But, the IMU is still not read. 

Could there be a problem with IMU itself? How could we check it? the only thing that confuses me, and makes me believe that it is working fine, as when I connect it with Matlab, it gives me some data as mentioned [here](https://github.com/LORD-MicroStrain/MSCL/issues/84#issuecomment-513721895) The detail of the serial port on Matlab is given at the end:
PS: I tried to change the USB port; thus, the name of the port is changed. 
```Serial Port Object : Serial-/dev/ttyACM0

   Communication Settings 
      Port:               /dev/ttyACM0
      BaudRate:           9600
      Terminator:         'LF'

   Communication State 
      Status:             closed
      RecordStatus:       off

   Read/Write State  
      TransferStatus:     idle
      BytesAvailable:     120
      ValuesReceived:     512
      ValuesSent:         0

rwslord, 2019-07-22:
Do you have access to a Windows machine so that we could try out SensorConnect or MIP Monitor to verify the device is working?

Even though you're getting bytes in Matlab, it's concerning because I'm not seeing any start of packet bytes like I would expect (0x75, 0x65 or 117, 101 decimal) in your attached file.

vibhoraggarwal, 2019-07-22:
I tried to run it on Windows using [MIP Monitor](https://www.microstrain.com/software#desktop). 

The software is unable to detect the IMU. Strangely at times, the IMU is shown in the device manager, but, still it does not show in the software. The device manager gives me an error and wants me to update the drivers, which are already updated to the website from Microstrain.

rwslord, 2019-07-22:
Ok, if you don't mind, please send an email to our Inertial customer service engineer at Barry_Trutor@lord.com. He should be able to troubleshoot the device not working properly in MIP monitor. Once we get that figured out, we should be able to get it working in MSCL.

vibhoraggarwal, 2019-07-22:
Thanks for such quick responses @rwslord I shall contact Barry Tutor; and, get back to you when he replies.

vibhoraggarwal, 2019-07-22:
I got a reply from Barry Trutor. 

> The 3DM-GX3-25 USB driver that we provide will not work in Windows 10.
It is an old driver and we stopped supporting it about 10 years ago.
A quick fix would be to get one of our RS-232 cables (6212-1000) and use an RS-232-to-USB adapter with it.
http://files.microstrain.com/mechanical-prints/4005-0037.pdf
That configuration will work with Windows 10, a USB port, and the 3DM-GX3-25.
Our distributor can provide the cable.

Could it be a good assumption why we are not able to connect it to Linux as well?

rwslord, 2019-07-22:
Hmm, I'm not sure. The 3DM-GX3-25 is definitely a legacy product, so it's possible it doesn't work on newer versions of Linux. Barry may be able to track down an answer for that question as well.

vibhoraggarwal, 2019-07-22:
I actually found a RS232 cable and the converter in the lab. I will test it in half hour from now, and will let you know.

vibhoraggarwal, 2019-07-24:
@rwslord 
I tested the IMU with the RS232 cable and the USB converter. 

It seems that the GUI software are indeed not compatible with Windows 10. 
I also checked it on Linux(Ubuntu 19.04) with the same setup and it does not run on it either.

rwslord, 2019-07-24:
@vibhoraggarwal  - Could you give a few more details on what you are seeing? 

Our PC software, MIP Monitor and SensorConnect, both support Windows 10. 

Did the RS232 cable work for you? or are you saying that it didn't work?

I would recommend contacting Barry Trutor again as well with any followup, as this seems related to hardware, and not MSCL.

vibhoraggarwal, 2019-07-24:
@rwslord
 > Could you give a few more details on what you are seeing?
> Did the RS232 cable work for you? or are you saying that it didn't work?

**RS232 cable did not work for me**. 
Although, differently from the previous application with just the USB cable, the IMU(COM 5) port in the device manager on the windows **does show that it is successfully installed**. But, on using the `MIP Monitor` or `3DM-GX3 Monitor` it is not shown as the connected devices.

I contacted Barry Tutor; this is what he said: 

> - Our USB driver for WINDOWS is not compatible with Windows 10.
 > - Our Monitor 1.7 Windows software (which is the software that we shipped with this unit) is not compatible with Windows 10. 

The only thing that confuses me is if the device is not compatible with the new linux and the Windows it should not even send the raw data after connecting; which it is sending. For example, I am attaching a log file which I received after tracking raw data using `Putty` on Windows.
[putty.log](https://github.com/LORD-MicroStrain/MSCL/files/3426755/putty.log)

rwslord, 2019-07-24:
Ok, from Barry's original response, it seem like an RS232 cable should allow the 3DM-GX3-25 to work properly with our MIP Monitor software.

I apologize that I can't help you further with this particular issue. Please feel free to work further with Barry to get your device working in MIP Monitor. Once it's working there, it should work in MSCL, and I can help with further MSCL questions if you have any.

mglord, 2021-06-15:
Sorry for the inconvenience with the GX3-25 - hopefully you were able to get everything working for your application with the GX4-25!

I'm going to go ahead and close this, but feel free to reach out if you run into any additional issues! Thanks!

closed 2021-06-15 by mglord