LORD-MicroStrain/MSCL #86: Wrong Coordinate System - CH_FIELD_ESTFILTER_ESTIMATED_LLH_UNCERT,

mnewcomb, 2019-09-03:
Device: 3DM-GX5-45
If I request CH_FIELD_ESTFILTER_ESTIMATED_LLH_UNCERT from the device, the returned channel names from MSCL are:
estNorthPositionUncert
estEastPositionUncert
estDownPositionUncert

These names are defined in MSCL/source/mscl/MicroStrain/MIP/MipTypes.cpp.  Given I've asked for position uncertainty in Latitude and Longitude and Height (LLH), I'm confused why MSCL is giving me position uncertainty information in the NED coordinate system (North / East / Down ).

Is this REALLY NED?

gpcarpenter, 2019-09-03:
Hi there, 

Gregg Carpenter here from the team that developed the 3DM-GX5-45. 

This is really NED, calculated and reported in terms of meters. Internal to the device all position uncertainties are handled relative to meters. This offers a couple of advantages both in terms of internal precision overhead, as well as consistency relative to absolute distance anywhere on the globe. Uncertainty in position relative to degrees isn't offered as a reported data quantity by the device.
 
https://www.microstrain.com/sites/default/files/3dm-gx5-45_dcp_manual_8500-0064_2.pdf

mnewcomb, 2019-10-13:
Okay!  Thank you for the manual pointer.  It's helpful!

closed 2019-10-13 by mnewcomb