LORD-MicroStrain/MSCL #87: How to build the examples on Ubuntu

zakjeens, 2019-09-26:
Hello!

I just downloaded the MSCL prebuilt and source versions. Installing it was no problem, however I am having a hard time understanding the directions for the examples.
I am using C++, so I copied all the header files from /usr/share/c++-mscl/source/mscl to dependencies/mscl
However I cannot find anything named MSCL.lib. Where should I be looking for that?
And how should I built the examples afterwards? What commands do I use? Doing "cmake ." and "make" didn't work.

---
A slightly related question. I am going to be using VLU850 tension sensors (8 of them, connected to two VLINK200 and a WSDA200 plugged to my pc) on my robot, which is controlled via ROS. Do you have any ROS templates?

mglord, 2019-09-26:
Hi!

MSCL.lib is only included in the Windows C++ package, for Linux you will just need to link to libmscl.so (in base c++-mscl folder) when compiling. You do not need the source version with the prebuilt version successfully installed. Once you have the sample code you should be able to use any compiler you prefer that supports C++ 11 - there is an example command using GCC (g++ command) on the 'How to use MSCL' page under Linux/C++.

The C++ instructions on the MSCL Examples Readme are primarily applicable to a Windows environment. The prebuilt MSCL package includes Boost so you should not need to build that separately, and it's likely you won't need OpenSSL at all. The example code should have all the file includes set correctly as long as they are located in the same directory as one another, just make sure you point your compiler to use the installed MSCL and Boost (see example command on 'How to...' page). You will have to update the COM_PORT and NODE_ADDRESS strings in the main.cpp example.

Additionally, for the WSDA-200 you will need the linux-specific driver to properly communicate with the WSDA (if you don't already have it). You can find the driver [here](https://github.com/LORD-MicroStrain/Drivers/releases/tag/linux-cp210x-0.2).
When properly installed, your WSDA-200 should show up in the connected devices list as a ttyUSB device. To confirm, in directory /dev run:
`ls | grep ttyUSB` 

Re: ROS - We do not currently offer any ROS packages/drivers. I know that there are a few independently developed drivers available for some of our inertial products, but I am not aware of any for our wireless products. MSCL should reduce the work required to get your device connected and publishing data - we would be happy to answer any questions you have along the way!

zakjeens, 2019-09-27:
Hi, thanks for your answer!

I tried building the Displacement example with g++ like this:
g++ -I/usr/share/c++-mscl/source -I/usr/share/c++-mscl/Boost/include/ getCurrentConfig.h parseData.h setCurrentConfig.h setToIdle.h startSampling.h main.cpp -o Displ_example -L/usr/share/c++-mscl -lmscl -lstdc++ -std=c++11

(actually with and without the explicit naming of local header files like getCurrentConfig.h)
I get this error:
```
/usr/share/c++-mscl/libmscl.so: undefined reference to `pthread_create'
/usr/share/c++-mscl/libmscl.so: undefined reference to `pthread_condattr_setclock'
/usr/share/c++-mscl/libmscl.so: undefined reference to `pthread_join'
/usr/share/c++-mscl/libmscl.so: undefined reference to `pthread_detach'
collect2: error: ld returned 1 exit status
```

Since it complains about the MSCL library itself, I am at a loss as to what I could do?

zakjeens, 2019-09-27:
That got solved, turns out I need to add -pthread and -lpthread to the g++ command.

However, I am now not so sure how to launch it. Launching it on it's own as ./program_name gives error
`./program_name: error while loading shared libraries: libmscl.so: cannot open shared object file: No such file or directory`

And then launching it with 
`LD_PRELOAD=/usr/share/c++-mscl/libmscl.so ./program_name`
Returns an error:
`sh: symbol lookup error: /usr/share/c++-mscl/libmscl.so: undefined symbol: pthread_create`

On another note: when trying to build the wireless example with g++, it complains that:
`enableDisableBeacon.h: In function ‘void enableDisableBeacon(mscl::BaseStation&)’:
enableDisableBeacon.h:30:10: error: ‘std::this_thread’ has not been declared
     std::this_thread::sleep_for(std::chrono::seconds(3));
`
You have a missing include. It can be fixed by adding 
`#include<thread>` 
to that header file.

zakjeens, 2019-09-27:
One more update from my side:
I needed to add a library path to my ~/.bashrc to get rid of the lookup error, but the program is still unhappy:

`sh: 1: pause: not found`

zakjeens, 2019-09-27:
finally had to comment out the line
system("pause");
to get rid of that problem.

However, I am lacking experience in setting up such sensors (that aren't plug&play), so I do not know what should be set in the COM_PORT and where I can find it? And similarly how can I find the NODE_ADDRESS? I saw that a number gets assigned in SensorConnect on Windows, but is there a way to know this number while on Ubuntu? Should I just launch my two VLINK200 while on Windows, record their values and use them as NODE_ADDRESS in this example code on Ubuntu?

zakjeens, 2019-09-27:
I found the function in your library to get the COM_PORT. However now it complains that:
`Error: Connection error.
`
This error description unfortunately does not give enough details for me to try to solve it. Could you please help?

And I still need the NODE_ADDRESS from previous post.

mglord, 2019-09-27:
Thanks for your patience with this and letting us know how you've been able to resolve the issues you're encountering - that will be super helpful in making process and documentation improvements moving forward!

I'll go back through and address everything more thoroughly in a little while, but a couple things that I should be able to help resolve quickly:

- For the COM_PORT string: try something like "/dev/ttyUSB0" (number may differ). You can run the ls command in my previous comment from the /dev directory to see all connected USB devices.
- The baud rate for the WSDA-200 is 3000000 - good call on including the baud rate when building the Connection object, I think that is required when running on linux environments.
- The NODE_ADDRESS value is set on the node itself and does not change between environments - the Node Address indicated in SensorConnect is the value you need. If you don't know the node address, you can get a list of discovered nodes from the WSDA with the BaseStation::getNodeDiscoveries function.

Again, I'll go back through everything shortly, but hopefully this helps for now!

zakjeens, 2019-09-30:
Hi,

Thank you for the answers! Indeed, setting the baud rate to 3000000 allowed me to communicate with the nodes.
However the node discovery does not work for some reason. I did like this:
```
mscl::BaseStation baseStation(connection);
mscl::NodeDiscoveries all_nodes = baseStation.getNodeDiscoveries();
std::cout<<"before for loop"<<std::endl;
for (unsigned int i=0;i<all_nodes.size();i++) {
      std::cout<<"found Node "<<i<<std::endl;
}
std::cout<<"after for loop"<<std::endl;
```
Nothing gets printed between the "before for loop" and "after for loop" terminal outputs. What am I doing wrong?
By the way, when I define the NODE_ADDRESS myself (as seen on SensorConnect), then I can connect and get my sensor readings easily.

------------

Just one last question, more like a confirmation really. On Windows I did all the calibration necessary to make the sensor measurements show up as Newtons in a predefined range. But when I launched the parseData function, the sensor readings were in bits. I edited the example so that:

```
mscl::WirelessNodeConfig config;
config.dataFormat(mscl::WirelessTypes::dataFormat_cal_float);
node.applyConfig(config);
//Example: Start Sampling
startSyncSampling(baseStation, networkNodes);
//Example: Parse Data
parseData(baseStation);
```
As I understand, the applyConfig function only changes what is defined in the `mscl::WirelessNodeConfig config` and does not touch anything else, right? The readings now seem to be correct, I just wanted to check with you if I am doing everything right.

mglord, 2019-09-30:
Regarding the BaseStation::getNodeDiscoveries function - I should have explained this more clearly above. When a node is powered on it sends out discovery packets over each frequency to let any listening base stations know it's there - this is why in SensorConnect a node will only be automatically detected if it is powered on while the application is open and the base station is plugged in. In order to detect nodes with getNodeDiscoveries() the node must be powered on after the BaseStation object is instantiated. For example:

`mscl::BaseStation bs(c);
bool nodeFound = false;
mscl::NodeDiscoveries all_nodes;

// power-on node while loop is running
while (!nodeFound)
{
	all_nodes = bs.getNodeDiscoveries();
	nodeFound = all_nodes.size() > 0;
}

for (unsigned int i = 0; i < all_nodes.size(); i++)
{
	std::cout << "found Node " << all_nodes[i].nodeAddress() << std::endl;
}`

Obviously this is a simplified example that will hang indefinitely unless a node is found, but hopefully it helps to illustrate how this is used. Fortunately, node addresses don't change so storing a list to attempt connections will also work if your use case requires nodes be on prior to running your application.

Regarding the WirelessNode::applyConfig function - yes, your understanding is correct. applyConfig will only update settings that were manually changed on the WirelessNodeConfig object. This can also be configured in SensorConnect from the Sampling screen by selecting 'float' in the Data Type column.

zakjeens, 2019-10-04:
Hi! Thanks for the reply, now I get it. Indeed in my case it is easier to just store the node addresses.

I have run into a synchronisation problem. I have 8 sensors that need to be read at the same time. They are connected to two nodes,that are working at max speed, meaning that nodes are set to 256Hz frequency and having 4+2 channels fills the network bandwidth to 100%.
As I understand, the only way to get sensor measurements is by using mscl::DataSweeps and one single sweep is from one node only. Is there a way to make sure that two consecutive sweeps are from different nodes?
In a very simplified example;
```
while(ros::ok()){
      mscl::DataSweeps sweeps = baseStation.getData(2);
      for(mscl::DataSweep sweep : sweeps){
        if(sweep.nodeAddress()==NODE_ADDRESS){
          std::cout<<"ONE !! time="<<sweep.timestamp().str()<<" data size="<<sweep.data().size() <<std::endl;
        }
        if(sweep.nodeAddress()==NODE_ADDRESS2){
          std::cout<<"TWO !! time="<<sweep.timestamp().str()<<" data size="<<sweep.data().size()<<std::endl;
        }
      }
}
```
And then this is what's printed:

```
ONE !! time=2019-10-04 13:12:14.960937500 data size=4
ONE !! time=2019-10-04 13:12:14.964843750 data size=4
ONE !! time=2019-10-04 13:12:14.968750000 data size=4
ONE !! time=2019-10-04 13:12:14.972656250 data size=4
ONE !! time=2019-10-04 13:12:14.976562500 data size=4
ONE !! time=2019-10-04 13:12:14.980468750 data size=4
ONE !! time=2019-10-04 13:12:14.984375000 data size=4
ONE !! time=2019-10-04 13:12:14.988281250 data size=4
TWO !! time=2019-10-04 13:12:14.968750000 data size=4
TWO !! time=2019-10-04 13:12:14.972656250 data size=4
TWO !! time=2019-10-04 13:12:14.976562500 data size=4
TWO !! time=2019-10-04 13:12:14.980468750 data size=4
ONE !! time=2019-10-04 13:12:14.992187500 data size=4
ONE !! time=2019-10-04 13:12:14.996093750 data size=4
ONE !! time=2019-10-04 13:12:15.000000000 data size=4
ONE !! time=2019-10-04 13:12:15.003906250 data size=4
TWO !! time=2019-10-04 13:12:14.984375000 data size=4
TWO !! time=2019-10-04 13:12:14.988281250 data size=4
TWO !! time=2019-10-04 13:12:14.992187500 data size=4
TWO !! time=2019-10-04 13:12:14.996093750 data size=4
TWO !! time=2019-10-04 13:12:15.000000000 data size=4
TWO !! time=2019-10-04 13:12:15.003906250 data size=4
ONE !! time=2019-10-04 13:12:15.007812500 data size=4
ONE !! time=2019-10-04 13:12:15.011718750 data size=4
ONE !! time=2019-10-04 13:12:15.015625000 data size=4
ONE !! time=2019-10-04 13:12:15.019531250 data size=4
TWO !! time=2019-10-04 13:12:15.007812500 data size=4
TWO !! time=2019-10-04 13:12:15.011718750 data size=4
TWO !! time=2019-10-04 13:12:15.015625000 data size=4
TWO !! time=2019-10-04 13:12:15.019531250 data size=4
TWO !! time=2019-10-04 13:12:15.023437500 data size=4
TWO !! time=2019-10-04 13:12:15.027343750 data size=4
ONE !! time=2019-10-04 13:12:15.023437500 data size=4
ONE !! time=2019-10-04 13:12:15.027343750 data size=4
ONE !! time=2019-10-04 13:12:15.031250000 data size=4
ONE !! time=2019-10-04 13:12:15.035156250 data size=4
```
It seems that every time the base station latches unto one node and keeps reading it for at least 4 sweeps and only then tries the other one. Any way to change this? I can reduce the frequency to 128Hz if that could help?

zakjeens, 2019-10-07:
Hi, I haven't gotten any replies yet, but I decided to check the example as well just to be sure that I didn't make an error on my side. Once I launch the         `parseData(baseStation);` function, I get this in the output:

```
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.027343750 Tick: 0 Sample Rate: 256Hz Base RSSI: -38 Node RSSI: -34 
DATA: ch1: 0.101482 ch2: -0.003872 ch3: -0.015823 ch4: 0.012352 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.027343750 Tick: 0 Sample Rate: 256Hz Base RSSI: -43 Node RSSI: -41 
DATA: ch1: 0.051144 ch2: 0.022472 ch3: 0.063351 ch4: -0.021515 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.031250000 Tick: 1 Sample Rate: 256Hz Base RSSI: -37 Node RSSI: -34 
DATA: ch1: 0.102215 ch2: -0.003872 ch3: -0.014694 ch4: 0.015232 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.035156250 Tick: 2 Sample Rate: 256Hz Base RSSI: -37 Node RSSI: -34 
DATA: ch1: 0.102215 ch2: -0.007744 ch3: -0.009041 ch4: 0.021408 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.039062500 Tick: 3 Sample Rate: 256Hz Base RSSI: -37 Node RSSI: -34 
DATA: ch1: 0.109543 ch2: -0.006886 ch3: -0.013187 ch4: 0.020996 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.031250000 Tick: 1 Sample Rate: 256Hz Base RSSI: -43 Node RSSI: -42 
DATA: ch1: 0.050735 ch2: 0.025017 ch3: 0.064564 ch4: -0.016548 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.035156250 Tick: 2 Sample Rate: 256Hz Base RSSI: -43 Node RSSI: -42 
DATA: ch1: 0.053600 ch2: 0.022049 ch3: 0.064964 ch4: -0.016548 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.039062500 Tick: 3 Sample Rate: 256Hz Base RSSI: -43 Node RSSI: -42 
DATA: ch1: 0.052784 ch2: 0.021198 ch3: 0.072632 ch4: -0.019447 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.042968750 Tick: 4 Sample Rate: 256Hz Base RSSI: -43 Node RSSI: -42 
DATA: ch1: 0.054829 ch2: 0.023743 ch3: 0.073841 ch4: -0.014893 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.046875000 Tick: 5 Sample Rate: 256Hz Base RSSI: -43 Node RSSI: -42 
DATA: ch1: 0.053600 ch2: 0.021198 ch3: 0.071827 ch4: -0.021099 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.042968750 Tick: 4 Sample Rate: 256Hz Base RSSI: -38 Node RSSI: -34 
DATA: ch1: 0.108810 ch2: -0.005161 ch3: -0.006027 ch4: 0.018940 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.046875000 Tick: 5 Sample Rate: 256Hz Base RSSI: -38 Node RSSI: -34 
DATA: ch1: 0.113571 ch2: -0.005161 ch3: -0.003391 ch4: 0.022644 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.050781250 Tick: 6 Sample Rate: 256Hz Base RSSI: -38 Node RSSI: -34 
DATA: ch1: 0.116135 ch2: -0.011185 ch3: -0.006783 ch4: 0.024292 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.054687500 Tick: 7 Sample Rate: 256Hz Base RSSI: -38 Node RSSI: -34 
DATA: ch1: 0.116505 ch2: -0.011185 ch3: -0.001881 ch4: 0.025116 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.058593750 Tick: 8 Sample Rate: 256Hz Base RSSI: -38 Node RSSI: -34 
DATA: ch1: 0.113937 ch2: -0.007744 ch3: -0.006405 ch4: 0.027172 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.050781250 Tick: 6 Sample Rate: 256Hz Base RSSI: -44 Node RSSI: -42 
DATA: ch1: 0.051144 ch2: 0.023319 ch3: 0.069809 ch4: -0.014893 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.054687500 Tick: 7 Sample Rate: 256Hz Base RSSI: -44 Node RSSI: -42 
DATA: ch1: 0.052784 ch2: 0.020351 ch3: 0.071018 ch4: -0.014481 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.058593750 Tick: 8 Sample Rate: 256Hz Base RSSI: -44 Node RSSI: -42 
DATA: ch1: 0.054829 ch2: 0.021198 ch3: 0.073841 ch4: -0.019447 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.062500000 Tick: 9 Sample Rate: 256Hz Base RSSI: -44 Node RSSI: -42 
DATA: ch1: 0.053600 ch2: 0.019081 ch3: 0.074245 ch4: -0.016964 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.066406250 Tick: 10 Sample Rate: 256Hz Base RSSI: -44 Node RSSI: -42 
DATA: ch1: 0.057693 ch2: 0.022049 ch3: 0.072227 ch4: -0.017376 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.062500000 Tick: 9 Sample Rate: 256Hz Base RSSI: -37 Node RSSI: -34 
DATA: ch1: 0.112473 ch2: -0.009895 ch3: -0.003014 ch4: 0.026764 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.066406250 Tick: 10 Sample Rate: 256Hz Base RSSI: -37 Node RSSI: -34 
DATA: ch1: 0.115036 ch2: -0.009895 ch3: -0.004143 ch4: 0.025528 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.070312500 Tick: 11 Sample Rate: 256Hz Base RSSI: -37 Node RSSI: -34 
DATA: ch1: 0.117237 ch2: -0.006886 ch3: -0.004520 ch4: 0.027584 
Packet Received:  Node 27405 Timestamp: 2019-10-07 13:08:43.074218750 Tick: 12 Sample Rate: 256Hz Base RSSI: -37 Node RSSI: -34 
DATA: ch1: 0.118702 ch2: -0.010757 ch3: -0.006027 ch4: 0.021820 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.070312500 Tick: 11 Sample Rate: 256Hz Base RSSI: -44 Node RSSI: -42 
DATA: ch1: 0.054008 ch2: 0.024593 ch3: 0.073441 ch4: -0.016548 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.074218750 Tick: 12 Sample Rate: 256Hz Base RSSI: -44 Node RSSI: -42 
DATA: ch1: 0.053192 ch2: 0.019081 ch3: 0.073441 ch4: -0.018204 
Packet Received:  Node 27404 Timestamp: 2019-10-07 13:08:43.078125000 Tick: 13 Sample Rate: 256Hz Base RSSI: -44 Node RSSI: -42 
DATA: ch1: 0.051556 ch2: 0.023319 ch3: 0.078686 ch4: -0.020275 
[... cut ...] d as a feedback for a controller, then the readings have to be as up to date as possible.

Do you have any suggestions on how to ensure a better behavior? If it would be possible to force only TWO consecutive sweeps to be from the same node, it would be perfect! What I mean by this is:

```
...
Node 27404 Tick: 10
Node 27404 Tick: 11
Node 27405 Tick: 11
Node 27405 Tick: 12
Node 27404 Tick: 12
Node 27404 Tick: 13
Node 27405 Tick: 13
Node 27405 Tick: 14
...
```

Is there a way to make this happen?

mglord, 2019-10-07:
As far as I'm aware there is no way to guarantee the order of sweeps/packets, but if you get multiple points for the same channel in one packet you can calculate an accurate timestamp based on the sample rate and the timestamps across nodes should be synchronized.

If you are more concerned about getting the data as soon as it's sampled/available in order to react to it right away you could look into using the low duty cycle sampling mode. The low duty cycle sampling mode is not synchronized so data is transmitted immediately after sampling but there is an increased risk of dropped packets in high traffic situations.

To initialize non-sync sampling via MSCL, you will need to do the following for each node:
- Configure the node for non-sync sampling by setting samplingMode on a WirelessNodeConfig object to WirelessTypes::SamplingMode::samplingMode_nonSync before passing it into WirelessNode::applyConfig()
- Start sampling by calling WirelessNode::startNonSyncSampling()

There should be no change to how the data is collected and read.

Let me know if this looks like it will work better for you - if not I can reach out to someone with more experience specifically in wireless network configuration to help find the best solution!

zakjeens, 2019-10-08:
I tried two things:
1. Reducing the Hz to 64Hz in the hopes that then the readings of different nodes might be closer to one another. The result is:
```
Node 27404 Tick: 10
Node 27404 Tick: 11
Node 27405 Tick: 10
Node 27405 Tick: 11
Node 27404 Tick: 12
Node 27404 Tick: 13
Node 27405 Tick: 12
Node 27405 Tick: 13
```

So unfortunately it is still not what I want.

2. I also tried the non-sync sampling, but I get the feeling that I am doing something wrong because only one node actually starts sampling. The first one to get called is the one that samples, while the other shows up at the very beginning of the sweeps for 1 or maybe 2 times and then goes to iddle without me setting it. I do not see any info about it being iddle in the terminal, but the light on the device changes to the iddle pattern.

```
    mscl::WirelessNodeConfig config;
    config.dataFormat(mscl::WirelessTypes::dataFormat_cal_float);
    config.samplingMode(mscl::WirelessTypes::SamplingMode::samplingMode_nonSync);
    //config.samplingMode(mscl::WirelessTypes::SamplingMode::samplingMode_sync);

    node2.applyConfig(config);
    node1.applyConfig(config);

   node2.startNonSyncSampling();
   node1.startNonSyncSampling();
```
In this example node2 is sampled with 64Hz rate and node1 goes to iddle after the very first sweep.
If i change the order of the last two lines, it will be node1 that is sampled, while the other node goes to iddle state.

What should I be doing differently? I cannot make a network, since only SyncSamplingNetwork exists and we need non-sync.

-----------

I also found that there is a sampling mode  called Sync Sampling Burst, where one can influence burst duration and the number of sweeps currently set on the node? Could this be a possible solution?

zakjeens, 2019-10-14:
Hello, it's been a week, could you please take a look at my previous post?

Am I trying the non-sync sampling in a wrong way? Is there a way to launch non-sync sampling from the baseStation and not for each node separately? I've found that in full documentation there's a function baseStation.node_startNonSyncSampling(), but it requires a WirelessProtocol object that I have no idea how to define.

And secondly - would burst sampling do the trick?

zakjeens, 2019-10-15:
The timing problem seems to be worse than expected. I checked the current time that is retrieved by mscl library and the sweep time, which turns out is almost a second old! This is unacceptable!
```
 std::cout<<"ONE !! time="<<sweep.timestamp().str()<<" data size="<<sweep.data().size()<<" Tick:"<<sweep.tick()<< " mscl curr time "<<mscl::Timestamp::timeNow().str()<<std::endl;
```
gives
```
ONE !! time=2019-10-15 08:53:09.984375000 data size=4 Tick:61 mscl curr time 2019-10-15T08:53:11.015500
ONE !! time=2019-10-15 08:53:10.000000000 data size=4 Tick:62 mscl curr time 2019-10-15T08:53:11.015536
ONE !! time=2019-10-15 08:53:10.015625000 data size=4 Tick:63 mscl curr time 2019-10-15T08:53:11.015569
ONE !! time=2019-10-15 08:53:10.031250000 data size=4 Tick:64 mscl curr time 2019-10-15T08:53:11.015602
```

Considering the high cost of the sensors and accompanying hardware, this is simply ridiculous!

jpherbst, 2019-10-15:
Hello,

If you have a pressing need for support, please contact our support staff instead of using this portal, +1(802)876-6546 or Andrew_Deering@LORD.COM.

There are a few configuration options for sampling our wireless nodes.

Synchronized Lossless (default):
Each node receives time from the base station periodically and is configured to transmit during specific slots during each second.  The node will buffer data until it is scheduled to transmit at which point it transmits as much data as will fit in the slot it is configured for.  The base station must acknowledge the receipt of the packet or the node will retransmit the packet in its next configured slot.  This method is not designed for low latency because depending on your node's configuration slots can be spaced out up to a second for a node and there may be many slots between acknowledged transmissions due to wireless interference.  However in this configuration nodes are synchronized together to +/- 50 us, there can be many nodes on the network, susceptibility to wireless interference is minimized, and power consumption is reduced by buffering data before sending.

Synchronized without Lossless:
Similar to _Synchronized Lossless_ except the node does not listen for acknowledgements from the base station, making it lower latency but it still waits for a slot to transmit.

Unsynchronized:
In this mode each node sends immediately after sampling, but does not timestamp data and no inter-node synchronization guarantees.  This mode is the lowest latency between sampling and data reception, but nodes can interfere with one another resulting in data loss and communication issues especially with sample rates greater than 1 Hz.

If you communicate your requirements to us we can help you determine the best configuration for your wireless network.

Jon

mglord, 2019-10-15:
As is addressed in Jon's response, non-sync sampling is prone to dropped packets and communication interruptions. I've been testing things out with a couple nodes here and have gotten similar results to what you are seeing. There is nothing wrong with your code to initialize non-sync sampling - baseStation.node_startNonSyncSampling() is utilized by the node to start each sampling individually. I suspect you are experiencing dropped packets due to high sample rates over an unsynchronized network.

I had some success utilizing an unsynchronized network by dropping the sample rate down even lower and also switching to a raw integer data format instead of floats. You can save bandwidth by getting the calibration coefficients (slope and offset) from the node prior to sampling with WirelessNode::getLinearEquation() and then applying them (calibrated value = (slope x raw integer) + offset) once your application receives the value. From some quick testing, I was seeing pretty consistent results sampling raw int24 data from 8 channels across two nodes at 16 Hz. What is the minimum frequency at which you need data input to your application?

From what I understand, burst sampling is unlikely to resolve this issue unless you do not need continuous data output for both sensors simultaneously. I would recommend using SensorConnect to test the behavior of different sampling configurations or reach out to support (contact info above) to help determine the best option for your specific use case.

Regarding the latency between the sampled time (sweep.timestamp()) and the cout time (Timestamp::timeNow()): keep in mind there are a number of things contributing to this delay, including data buffering for synchronization and the time it takes to write a line for each sweep to the console. I recommend we get your network configuration set and running prior to addressing latency concerns as the issue may be resolved with the proper setup.

Let me know if you continue to see the issue where only one node samples upon initializing a non-sync network after reducing bandwidth (reduce sample rate, change data type, reduce channels, etc.)!

zakjeens, 2019-10-16:
Jon, thank you for the contact info. Unfortunately, there's no actual support or advice coming from that contact, at least judging by the first response I got.

mglord, about the frequencies that we need:
The robot works at 500Hz, but of course no one expects that from these sensors. I would say 128Hz would be good, or 64Hz in worst case. However it is also the latency that is important here. There's no point in working at 128Hz if the received data is old.
From what I've seen frequence and latency wise - I would prefer setting the non-synchronised mode and getting both nodes to work at the same time.
Thus my question - **if we would buy another WSDA200 and connect each node to its own WSDA200, could I get the desired behavior by setting each of them to non-sync mode?** I would then have to deal with synchronisation of data on my own. I wonder if mscl library has any way of comparing two timestamps or accessing that time as seconds?

And thank you for the idea of using raw integer data. Indeed, I can apply calibration coefficients after receiving it, if it helps with reducing latency!