LORD-MicroStrain/MSCL #88: WSDA-200-USB Com Port Help

Testing1User, 2019-10-15:
Hi,

Currently have your mscl library compiled in a raspbian OS. Works perfectly. 
However, I have 3 other usb inputs and the WSDA-200-USB does not always show up as /dev/tty0 (goes to tty1 or tty2). tty0 being the com port I specify in my scripts. 

I can't change the com port line in my script all the time. Therefore, I have tried setting the WSDA-USB-200 to a static address using [UDEV ](https://wiki.archlinux.org/index.php/Udev#Writing_udev_rules)rules found here.
The scripts do not work when I use this method. 

Summary: Trying to set my master file to have - dev/*custom_name* which links to the WSDA-USB-200 rather than change the dev/ttyUSB0 everytime.

Any info or a little guidance would be amazing please.
Cheers,

jpherbst, 2019-10-16:
Here's an article from SiLabs on the subject:
https://www.silabs.com/community/interface/knowledge-base.entry.html/2016/06/06/fixed_tty_deviceass-XzTf

I haven't tried this out myself, will take a look at it in a few days, but let us know if you get it to work.

Another potential solution is to look up the tty device through the usb subsystem in the sysfs, but that's probably more work than sym-linking with udev.

Testing1User, 2019-10-16:
Hi jpherbst,

That is the method I had used previously and I get "Invalid com port" error from ping node example script if I try to change my com port to the new assigned one.

bakulapal, 2019-10-18:
Hi,
For some reason udev symlinks are not working in mscl. If you use python then try this workaround:
COM_PORT = os.path.realpath("dev/custom_name")

Testing1User, 2019-10-18:
Hi bakulapal,

This does the job.
Thank you !

closed 2019-10-18 by Testing1User