LORD-MicroStrain/MSCL #89: Node Channel per mscl enable/disable

BaldeepSinghAtMack, 2019-10-17:
[image]

all i want ist to manipulate the channel Active check box set/reset

please see attached pic

i tried following

foreach (var group in ArrayMyNodes[i].MSCL_MyNode.features().channelGroups())//iterate over each channel group
                            {
                                for (int m = 1; m < ArrayMyNodes[i].myNodeChannels.Length; ++m)
                                {
                                    group.channels().enable((byte)(m), false);

                                }
                            }

here, i tried to activate each channel....with

mglord, 2019-10-17:
If I'm understanding correctly, I think what you're looking for is WirelessNode.features().channels().
Calling features().channels() will return a list of each channel with an associated ID, type, name, etc.
Calling features().channelGroups() returns a list of groups of channels that share configuration settings. For example, the RTD-Link-200 has many channels that share some settings so channelGroups() would return a list something like this:
- Channels: 1, 2, 3 (represented by a ChannelMask), Settings: Temperature Sensor, Input Range
- Channels: 4, 5, 6, Settings: Temperature Sensor, Input Range
- Channels: 1, 2, 3, 4, 5, 6, Settings: Low Pass Filter
This indicates which groups of channels share which settings.

Calling group.channels() (ChannelGroup.channels()) will return a ChannelMask object which is a group of channels represented by a bit mask. ChannelMask.enable(channelId, bool) will add or remove the provided channel ID from the channel mask based on the the boolean value provided (true - add channel, false - remove).
ChannelMask.enabled(channelId) will check whether or not the provided channel is included in the channel mask.

I would not recommend updating any values in a NodeFeatures object (WirelessNode.features()) - once initialized, it provides information about what is supported by the node's hardware and firmware. If you change these values (group.channels().enable(id, bool)) the NodeFeatures object will no longer accurately represent the supported features.

If you just want to iterate through all the channels on the node and check the box in the UI, all you need is the following:
```
foreach(var channel in node.features().channels()) // iterate over each channel
{
	// add logic to update UI checkbox here
	// you can identify the channel by channel.channelNumber() or channel.id()
}
```
If you want to check the 'Channel Enabled' box only under certain conditions, please provide more details about what you're trying to accomplish and I can help you figure out how to find the information you're looking for!

BaldeepSinghAtMack, 2019-10-18:
Dear Sir,

i try to clear the things again.
we have an MSCL based Application which does DataLogging with following Wireless Nodes :
1. V-Link
2. G-Link
Now, at different V-link channels inputs, we attach different Sensors.
i try to elaborate the situation with an example:-
it is sometimes wished that channel 1 stays active, channel 2 is inactive as nothing is connected and then 3 & 4 are active. Now My colleague starts the Node commander and disables the channel 2 so that in measurement file there are no irrelevant measurements for channel 2.
what i try with my application -> i have an ini file in which the Setup and Node configuration is saved and then after reading this file i configure the Node approprietaly.
I want to when in my ini file the "Channel 3 = inactive", the Node should not do any measurement on channel 3. as shown in picture, the channels 1, 2, and 4 are active and channel 3 is disabled.
I want to do this enabling and disabling with my application

[image]

Thanks and regards

BaldeepSinghAtMack, 2019-10-18:
i got the answer
i should change it with Node config

`mscl.ChannelMask activeChannels = new mscl.ChannelMask();
                            for (int ChNr = 0; ChNr<8; ChNr++) 
                            {
                                activeChannels.enable((byte)ChNr, Turn_1_True_0_False(MyIni.Read("CH" + ChNr + "_active", ArrayMyNodes[i].NodeName)));
                            }
                            config.activeChannels(activeChannels);
                            //ArrayMyNodes[i].MSCL_MyNode.applyConfig(config);`

 __

mglord, 2019-10-18:
Sorry for the initial misunderstanding and thanks for clarifying - I'm glad you were able to find the solution!

I'm going to go ahead and close this issue, but feel free to re-open it if you have any outstanding questions regarding this issue or reach out again if you run into any other problems!

closed 2019-10-18 by mglord