LORD-MicroStrain/MSCL #90: Pre-built MSCL package not working for raspbian

nayanagg, 2019-10-17:
Hi,

I tried to install the pre-built MSCL package(python3) in Pi 3A+. But it didn't work.

Help!

mglord, 2019-10-17:
Can you provide any additional details? What steps of the installation process did you get through? What is not working as expected - are there any error messages or unexpected behaviors?

Any additional information you can provide would be helpful.

nayanagg, 2019-10-18:
[image]

[image]
[image]

mglord, 2019-10-18:
When you installed the package, a folder (python3-mscl) with two files (mscl.py, _mscl.so) should have been added to usr/share. These files need to be moved to a directory where python looks for modules, or imported using their full file path.

You can find more detailed info here in the Linux/Python section: [How to use MSCL](https://github.com/LORD-MicroStrain/MSCL/blob/master/HowToUseMSCL.md)

Let me know if you continue to see the issue after moving the files!

nayanagg, 2019-10-19:
Hi,
issue resolved..!!
thanks

closed 2019-10-21 by mglord