LORD-MicroStrain/MSCL #91: Real time streaming

BaldeepSinghAtMack, 2019-10-18:
How do we do program real time streaming on a node?

i tried with start_nonsyncSampling but then i dont know how to get the sweeps?

mglord, 2019-10-18:
Wireless data collection is the same for both sync and non-sync networks - calling BaseStation.getData() on your BaseStation instance will return a collection of DataSweeps that have been buffered since the last time getData() was called. If you have multiple base stations, getData() will need to be called from each of them.

I would recommend looking through some of our example code for more context, particularly the parseData file.
I think in a previous issue you mentioned that you're using C# - those examples can be found here: [https://github.com/LORD-MicroStrain/MSCL/tree/master/MSCL_Examples/Wireless/CSharp/MSCL_Wireless_Example_CSharp/MSCL_Wireless_Example_CSharp](https://github.com/LORD-MicroStrain/MSCL/tree/master/MSCL_Examples/Wireless/CSharp/MSCL_Wireless_Example_CSharp/MSCL_Wireless_Example_CSharp)
Example code in other languages can be found from the root MSCL_Examples folder.

I don't think we have example code for starting a non-sync network, but if you are having trouble initializing sampling let us know and we can help you sort it out! It's usually helpful if you can provide the relevant section of code in order to identify any issues.

mglord, 2021-06-15:
Just an important note if anyone stumbles on this again looking for info: If you configure multiple nodes on the same frequency for non-sync sampling it is very likely you will experience dropped packets as the nodes try to transmit data at the same time - it is my understanding that if two nodes are trying to talk to the base station at the same time it won't hear either of them. Non-sync sampling over the same frequency for multiple nodes can be successful at lower sampling speeds, but to sample at high speeds you'll have the best luck getting a base station for each node and set each pair to a different frequency.

I'm going to close this, but if you run into any issues with implementation or need clarification on anything let us know!

closed 2021-06-15 by mglord