LORD-MicroStrain/MSCL #93: Empty packets via getDataPackets() with default timeOut

nayanagg, 2019-10-20:
[image]
[image]
[image]
[image]

**HELP!!**

mglord, 2019-10-21:
Please provide more information about the issue - what exactly is not working as expected?

nayanagg, 2019-10-21:
Hi,

As I've set a sample rate at 1khz. then the time taken for iterating 1000 readings should be 1 sec. but as you can see I'm getting 1000 readings in 1.7 sec which almost near to 500hz. So, kindly let me know how to get exact 1khz readings.

mglord, 2019-10-22:
It seems like this is a device configuration issue rather than an MSCL issue - MSCL can only collect and process packets as quickly as the device outputs them. I would encourage you to reach out to product support at: sensing_support@lord.com

If you have access to a Windows machine to help troubleshoot the issue it would be very helpful to install SensorConnect, our desktop application for configuration and data collection, to gather the following information and get a better understanding of the issue:
- Export device settings/configuration file
- Collect some sample data and export it to CSV so that we can more easily see the data frequency
Directions to obtain these vary by product line, but we can definitely provide further assistance if needed.

The installer and documentation/guides for SensorConnect can be found here: [https://www.microstrain.com/software/sensorconnect](https://www.microstrain.com/software/sensorconnect)

Also, what device(s) are you working with?

I would strongly encourage you to reach out to support so they can better assist you in troubleshooting sensor configuration. It will be helpful to them if you can provide all the above-requested information along with as much detail about the issue as possible.

If you do not have access to SensorConnect for troubleshooting let me know and we'll figure out another way to get the required information.

nayanagg, 2019-10-26:
It was baud rate issue with raspberry pi.
For 1000 Hz sample rate, baud rate must be define 921600 for both sensor and raspberry pi.

closed 2019-10-26 by nayanagg