LORD-MicroStrain/MSCL #95: Data not being returned in configured data type

DerekRuff, 2019-10-28:
I am using the G-Link-200 and MSCL to grab data from it. When I configure the G-Link I am setting it to store values as int16's. After that when I look at the data using the loggedDataSweep function data().as_int16() it returns incorrect values. When I pull it out using data().as_int32() all of the data looks good again.

mglord, 2019-10-30:
It looks like int16 data sent from a 20-bit node (like the G-Link) are truncated and shifted right prior to sending to reduce data size, so when reading the value from the packet we automatically shift it back and store it in an int32 to get an accurate value. WirelessDataPoint.as_int16() is returning an incorrect value because it is truncating the int32 to an int16 - because the value is not shifted prior to casting, the as_int16() function should not be used in this way.

You can check the stored data type of a data point by checking WirelessDataPoint.storedAs() - this will tell you exactly which format function should be used to read the data from the point. G-Link int24 data is also cast to int32 when processed, but is not truncated so bit shifting is not required.

To the function to determine what incoming data formats will be cast to is dataTypeToValueType() in [WirelessTypes.cpp](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL/source/mscl/MicroStrain/Wireless/WirelessTypes.cpp)
To see exactly how int16 data is processed when downloaded from the data log, check out parseNextSweep() in [DatalogDownloader.cpp](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL/source/mscl/MicroStrain/Wireless/DatalogDownloader.cpp)

Let me know if any of this is unclear or if you have additional questions!

DerekRuff, 2019-10-31:
Thanks, this answers my question.

closed 2019-10-31 by DerekRuff