LORD-MicroStrain/MSCL #99: WSDA-200-USB symlinks not working

zakjeens, 2019-11-14:
As stated in a [previous issue](https://github.com/LORD-MicroStrain/MSCL/issues/88), the symlinks do not work.
I have two WSDA-200-USB that can be connected in any order to any port. I set up udev rule like in [example here](https://www.silabs.com/community/interface/knowledge-base.entry.html/2016/06/06/fixed_tty_deviceass-XzTf), but I get the "Error: Invalid Com Port." at the line:
`
    mscl::Connection connection = mscl::Connection::Serial(COM_PORT, 3000000);
`
where `  COM_PORT = "/dev/lord-sensor-a";` leads to a symlink.

Is there any way around this? Or any fast way to check which discovered WSDA is which?
I'd like to avoid making a connection to both discovered WSDAs to check their frequencies to be able to distinguish them, but for now it seems the only option?

For me, each WSDA is on a different frequency with one node (and 4 sensors). Thus depending on which WSDA it is, I would give a different `NODE_ADDRESS`.

mglord, 2019-11-15:
I haven't had a chance to try it out but, based on the resolution of issue #88, it sounds like MSCL requires the absolute path (realpath) of the symlink:

> For some reason udev symlinks are not working in mscl. If you use python then try this workaround:
> COM_PORT = os.path.realpath("dev/custom_name")

This is the python workaround, but it looks like there may be a C++ version of the realpath function. Otherwise I'm sure there is some other way to get the absolute path of the symlink. I can try this out/look into this further on Monday.

Re: otherwise detecting which base station is which - when you get the list of connected BaseStations (mscl::Devices::listBaseStations()), each entry has a com port string key and the associated DeviceInfo object value has the device's serial number - if the same base station is always on the same frequency, you could try mapping the serial number to the frequency and associated node address and only connect to the one mapped to the desired frequency.

Again, I'll look into the symlink issue further on Monday, but in the mean time I hope this helps!

mglord, 2019-11-18:
I was able to resolve and utilize a symlink (udev rule) by using the [realpath](http://man7.org/linux/man-pages/man3/realpath.3.html) function to determine the actual port at run time.

Adding the following worked for me, let me know if this workaround doesn't resolve the issue!

Add to includes:
`#include <stdlib.h>`

Determine actual port:
```
const string COM_PORT = "/dev/bestBaseStation";
const int BAUD = 3000000;

char* actualPort = realpath(COM_PORT.c_str(), NULL);
string portString(actualPort);
cout << "Attempt create connection: " << COM_PORT <<" (" << portString << "), " << BAUD << endl;

//create a SerialConnection with the COM port
mscl::Connection connection = mscl::Connection::Serial(portString, BAUD);
cout << "Connection created" << endl;
```

Output:
```
Attempt create connection: /dev/bestBaseStation (/dev/ttyUSB1), 3000000
Connection created
```

zakjeens, 2019-11-20:
Thank you for the workaround, it solves the issue.

closed 2019-11-20 by zakjeens