LORD-MicroStrain/MSCL #100: setToIdle timeout

hpk-roquette, 2019-11-27:
Hi,

Using an Inertial node, from time to time the setToIdle() function times out (Error_Communication). This appears to be more frequent now that we request more data from the node. Even with:
```
auto old = node_->timeout();
node_->timeout(2000);
node_->setToIdle();
node_->timeout(old);
```
First of all, is this normal ? Should we be raising the timeout value higher ?

Because this fails from time to time, can we use the resume() function still ? Wondering because of its documentation: 
> Places the Node back in the mode it was in before issuing the setToIdle command.  If the setToIdle command was not issued, then the device is placed in default mode.

What happens if setToIdle previously failed ? What is default mode ?

Also, in our case the node will be used for a few hours at a time maximum (probably always at max frequency). In the majority of this time, we are reading every packet and saving them. Is it really useful to set the node to idle mode in between sampling sessions ? This line in the documentation says so:
> Unlike a Wireless Node, an Inertial Node can be directly communicated with even if it is sampling. However, in most cases, it is still useful to set the Node to an idle state between sampling sessions and when changing configurations.

Thanks a lot

hpk-roquette, 2019-11-27:
Bonus question if it's ok to add here: What is a valid data point ? Say for a MipDataPoint, what determines if it is valid or not ? Is it based on the number of satellites in view ? Is it different for each data point ?

mglord, 2019-12-02:
There is a lot of information in the product DCP (Data Communications Protocol document) that I think may help clarify some of these issues - I just wanted to bring this up first in case you are not familiar with this document to avoid confusion with the below answers/explanations. You can find the DCP on each product page under Documentation. Product page urls are structured: https://www.microstrain.com/inertial/<product name> (Ex: https://www.microstrain.com/inertial/3dm-gx5-45).

**Timeout error on setToIdle():**
This is likely caused by bandwidth limitations over the connection, particularly since you mentioned increased frequency of failure when streaming more data.
Unfortunately, if packets are being dropped due to bandwidth limitations, increasing the timeout will not resolve the issue because either the set to idle command is dropped and is never received by the device, or the device successfully receives and processed the set to idle command but the success response from the device is dropped. In either scenario it seems like the easiest way to ensure the command's success is to resend it on timeout until it succeeds.
There is more information about managing connection bandwidth in the DCP (usually section 2.8: Communications Bandwidth Management).

**Using resume():**
In my experience and testing the Resume command (0x01, 0x06) has always resumed streaming, even if Set to Idle was sent multiple times, and will not stop streaming if sent while streaming is enabled or Set to Idle failed. I believe the default mode is streaming enabled.

**Idle between sampling sessions:**
As far as I know, setting the device to idle between sampling sessions is simply recommended to avoid bandwidth difficulties when configuring or otherwise interacting with the device. Especially when streaming high volumes of data, if you need to communicate with the device between sampling sessions setting it to idle will reduce the chance of communication failures like what you are currently experiencing with the Set to Idle command.

These commands are outside of the scope of MSCL - MSCL handles sending the command to the device, but the actual behavior is dependent on the device firmware - so I don't have super in-depth knowledge of intended behaviors and configuration options to mitigate issues like this. For more information about these or other device behavior issues you can reach out Barry in product support (Barry_Trutor@LORD.COM) and he should be able to help you!

**Data point validity:**
The validity of a data point (MipDataPoint) is determined by the Valid Flags value that is included in the associated MIP field.
For example, the NED Velocity field (0x81, 0x05 - GNSS data set) includes the following values:
- North, East, and Down velocities
- Speed
- Ground Speed
- Heading
- Speed and Heading accuracies
- Valid Flags

The Valid Flags value is a bitfield that has different bits turned on/off to indicate the validity of the other values in the NED Velocity field, for example:
- 0x003F - All other values valid
- 0x0028 - Heading and Heading Accuracy values valid, all others invalid
	
Not all MIP data fields include validity information - MipDataPoint::hasValidFlag() should always be checked to determine whether MipDataPoint::valid() is meaningful rather than just defaulted to true.

More information about each data field and their associated Valid Flag mappings can be found in the DCP.

hpk-roquette, 2019-12-03:
Thanks a lot for the detailed response. I will be looking into that. Appreciate it.

closed 2024-10-24 by github-actions[bot]