LORD-MicroStrain/MSCL #101: Error: Connection error while using MSCL_Examples with C++

vibhoraggarwal, 2019-11-28:
Device: **3DM-GX4 -25**
System: **Ubuntu 18.04.3 LTS**
Problem: I am trying to run the [c++ examples](https://github.com/LORD-MicroStrain/MSCL/tree/master/MSCL_Examples/Inertial/C%2B%2B/MSCL_Inertial_Example_C%2B%2B/MSCL_Inertial_Example_C%2B%2B), but I am receiving the following error:
`Error: Connection error` 

I followed the issue #87, but in vain.

Issues already checked: 

- [x] Built libraries using MSCL libraries
- [x] Compiling the main file using the command: `g++ -I/usr/share/c++-mscl/source -I/usr/share/c++-mscl/Boost/include main.cpp -o main -L/usr/share/c++-mscl -lmscl -lstdc++ -std=c++11 -pthread`
- [x] Running the executed main file using the command `LD_PRELOAD=/usr/share/c++-mscl/libmscl.so ./main`
- [x] Creating connection using the all the baud rates mentioned [here](https://www.microstrain.com/support/faq/what-baud-rates-can-3dm-gx3-be-operated)
- [x] Locating and mentioning the correct COM_PORT using `ls /dev/` command 
- [x] Checked the IMU by running on the Windows OS on the same PC using the [Sensor connect](https://www.microstrain.com/software#desktop) and the IMU was successfully connected and displaying the data

mglord, 2019-12-02:
It sounds like you've double checked and accounted for the usual issues, thanks for including a clear list of what you've tried!

Off the top of my head I don't know what else is likely to be causing this issue - do you know which function/line is throwing this error? It may be helpful to see exactly how you're instantiating the Connection and InertialNode objects and/or the console output when you run the examples.

vibhoraggarwal, 2019-12-02:
At the following line is the error given:

`mscl::Connection connection = mscl::Connection::Serial(COM_PORT,115200);` 

So, the program doesn't even move ahead to use other MSCL functions.

mglord, 2019-12-02:
What is the value of COM_PORT?

vibhoraggarwal, 2019-12-02:
`/dev/ttyACM1`

mglord, 2019-12-02:
Thanks - that looks good, just wanted to double check it was formatted correctly.

Would you mind adding the following catch before the existing one in main.cpp to expose the error code?
```
catch (mscl::Error_Connection& e)
{
    cout << "Error: " << e.what() << " (" << e.code() << ")" << endl;
}
```
When run with this additional catch, what is the output/error code? It should either map to a standard linux system error code or it is a custom code that MSCL is throwing - either way it should help track down what is happening.

vibhoraggarwal, 2019-12-02:
Thanks for your suggestion.
 I will check which error it displays. I am at my home now, will be going to the lab in the morning. Then, I could tell about it more.

vibhoraggarwal, 2019-12-03:
> Would you mind adding the following catch before the existing one in main.cpp to expose the error code?
> 
> ```
> catch (mscl::Error_Connection& e)
> {
>     cout << "Error: " << e.what() << " (" << e.code() << ")" << endl;
> }
> ```

I checked the error the code returns with the above function you mentioned. Following is the error it returns:
`Error: Connection error. (13)`

mglord, 2019-12-03:
Based on a couple Linux system error code lists I found it looks like error code 13 is 'Permission Denied' - can you make sure your user is set up with permissions to access the device and retest?

vibhoraggarwal, 2019-12-03:
> Based on a couple Linux system error code lists I found it looks like error code 13 is 'Permission Denied' - can you make sure your user is set up with permissions to access the device and retest?

@mglord Thanks a lot. That was indeed the problem. 
We can also close the issue now, if you don't have something else to comment.
 
--
**For someone following the issue:** 

I used the following command to give the user to access the `/dev/ttyUSB[]` and `/dev/ttyACM[]` devices, as per the issue discussed [here](https://askubuntu.com/questions/112568/how-do-i-allow-a-non-default-user-to-use-serial-device-ttyusb0)

`sudoedit /etc/udev/rules.d/50-myusb.rules` 
and then saving the file with the following text: 
```
KERNEL=="ttyUSB[0-9]*",MODE="0666"
KERNEL=="ttyACM[0-9]*",MODE="0666"

closed 2019-12-03 by mglord