LORD-MicroStrain/MSCL #102: how to see VLINK200 battery status

zakjeens, 2019-12-18:
Hello,

Using SensorConnect on Windows I can see the "low battery" symbol on my nodes. This explains why I couldn't get measurements from these nodes. However I am working on Ubuntu and I can't seem to find a way to check node (VLINK200) battery status via the mscl library?

More precisely, I did find that in WirelessTypes.h there is BatteryStatus (and ExternalPowerIndicator for external power that I want to switch to), but I do not see how to use it.

mglord, 2019-12-19:
Hi -

Low battery information is transmitted from each device within each diagnostic packet and will be included as a WirelessDataPoint in the DataSweeps returned from BaseStation::getData() at the configured intervals (WirelessNode::getDiagnosticInterval() or can be viewed on the Sampling tab of the Configuration screen in SensorConnect).

SensorConnect detects low battery diagnostic data points by the channel ID (WirelessChannel::ChannelId::channel_diag_lowBatteryFlag) and updates the visual indicator based on the value. The value is a uint8 (WirelessDataPoint::as_uint8()) and should be one of the WirelessTypes::BatteryStatus enum values.
- 0 - battery good
- 1 - battery low
- 2 - battery critical

Hope this helps!

zakjeens, 2019-12-20:
Hi, 

For some reason when external power is connected, it drops the battery status message?

```
   mscl::ChannelData diagnostic = node.getDiagnosticInfo();
    for (unsigned int i=0;i<diagnostic.size();i++) {
      if(diagnostic[i].channelId()==mscl::WirelessChannel::ChannelId::channel_diag_lowBatteryFlag){
        std::cout<<"battery="<<+diagnostic[i].as_uint8()<<std::endl;
      }
      if(diagnostic[i].channelId()==mscl::WirelessChannel::ChannelId::channel_diag_externalPower){
        std::cout<<"external power="<<+diagnostic[i].as_uint8()<<std::endl;
      }
    }
```

i get battery=0 (status being "good") and external power=1.
I am guessing it stops checking the batteries once external power is available.

A security question: should I remove the batteries when using external power? I wanted to keep a fresh set of batteries just in case the external power stops (which can happen if the robot is stopped with emergency stop), however i am not sure. If external power stops, will the node start using battery power? And if external power restarts, will the node switch to external power or keep using battery?

ardLORD, 2019-12-20:
Leaving the batteries in while externally powered will allow the device to switch to battery when external power is lost (and back to external when it returns).  There will be very little to no drain on the batteries when the device is externally powered, and the switch over will be seamless (no data loss).

closed 2020-01-26 by zakjeens