LORD-MicroStrain/MSCL #103: Wired connection instead of WSDA+VLINK

zakjeens, 2020-01-26:
Hello,

I am not sure if this question should go here or not, to be honest. Please let me know.

I am using 2 WSDAs with one VLINK per WSDA and 4 sensors per vlink (VLU850 tension sensors).
While using the wireless setup is very much amazing at the lab, it unfortunately will fail completely during an exhibition our robot has to attend (because there are going to be countless other devices using wireless for communication). Unfortunately we have no way to test in advance which channels could be more or less usable, so we need to create a solution where we have a wired connection to the sensors.
Thus my questions: 
- is it possible to have a wired connection to the VLINKS? 
- Or maybe the sensors could be directly plugged into our industrial pc? 
- Would the MSCL library still be useful in the latter case? 
- And if not - can you suggest how we could get sensor readings from a wired connection?

ardLORD, 2020-01-27:
Hello,
It is not possible to hard wire wireless nodes.   You can try using an extension cable for the antenna of base to get it as close to the V-Link as possible.  Also direct connecting the V-Link antenna connection to the base antenna connection  is doable, but the transmit power on both base and V-Link need to be set to 0 dBm or damage will occur to the radio boards.

If you have further questions about hardware feel free to directly contact me at Sensing_support@lord.com

zakjeens, 2020-01-27:
Thank you, I've sent you an email.

closed 2020-01-27 by zakjeens