LORD-MicroStrain/MSCL #105: Get a single channel data from WirelessNode

alexandre2297, 2020-01-29:
Hi,

I'm using WirelessNodes and I would like to get a simple data for test my channels on this node.
In addition, I tryed to use function getDiagnosticInfo() but my nodes don't support it and I don't under
So, I would like to have sames features that the button "sample channel" on node Commander for having A/D value or calculted value.

Thanks a lot !
Best regards

mglord, 2020-02-10:
Hi - what node model(s) are you using? Different sensors support different functionality so it will be useful to know what you're working with in order to help you find the best solution.

alexandre2297, 2020-02-11:
Hi, this is my node model : G-Link2 2G-Internal

mglord, 2020-02-11:
**For configuring and initializing sampling, and parsing data** please refer to our example files located here:
[Wireless Examples](https://github.com/LORD-MicroStrain/MSCL/tree/master/MSCL_Examples/Wireless)
The setCurrentConfig, startSyncSampling, and parseData files should be helpful. Let me know if you have any specific issues or questions regarding implementation.

To set the node to sample only a subset of channels, the following configuration change is needed:
```
// build channel mask for channels you want enabled
mscl::ChannelMask channels;

// enables channels 1 and 3, change to whichever channels you want
channels.enable(1);
channels.enable(3);

// build config object with active channels and apply to node
mscl::WirelessNodeConfig config;
config.activeChannels(channels);
node.applyConfig(config);
```

**Regarding diagnostic info**: it looks like the diagnostic info available for the G-Link2 is limited to node and base RSSI information. @ardLORD - please correct me if this is inaccurate.
Node and base RSSI information is included in each DataSweep returned from BaseStation.getData().

**The Sample Channel functionality in Node Commander** uses a legacy command that is not implemented in MSCL. To achieve similar functionality you could start sampling and read out the channel's values to check against before setting the node back to idle. This will be a slower operation, but should provide more accurate results.

Hope this helps!

closed 2024-10-24 by github-actions[bot]