LORD-MicroStrain/MSCL #106: Unable to send PPS pulse from the GNSS receiver to IMU

vibhoraggarwal, 2020-02-10:
I am using [ublox EVK-6T-0-***](https://www.u-blox.com/sites/default/files/products/documents/EVK-6_ProductSummary_%28GPS.G6-EK-10031%29.pdf) as a GNSS receiver to send PPS pulse to [3DM-GX5-25 AHRS](https://www.microstrain.com/inertial/3dm-gx5-25) to synchronize them based on time. 

I am sending the GNSS time signals using the `getGNSSAssistTimeUpdate` command mentioned here: http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Inertial/InertialNode.h:InertialNode.getGNSSAssistTimeUpdate
 
Following is the snippet of the code, basically exploting the code in [parseData.h](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL_Examples/Inertial/C%2B%2B/MSCL_Inertial_Example_C%2B%2B/MSCL_Inertial_Example_C%2B%2B/parseData.h): 

```
...
static void parseData(mscl::InertialNode& node)
{
    //endless loop of reading in data
    while(true)
    {
        //get all the data packets from the node, with a timeout of 500 milliseconds
        mscl::MipDataPackets packets = node.getDataPackets(500);
        status.gnssPowerStateOn(1); 
        mscl::TimeUpdate timeupdate = node.getGNSSAssistTimeUpdate();
...
```
but it shows the error that the function is not supported, viz: 
`Error: The command (3108) is not supported.`

P.S: I might be doing something wrong too.

mglord, 2020-02-10:
The GNSS Assist Time Update command (0x0C, 0x24) is not supported on the GX5-25 - all supported commands can be found in the DCP here: [GX5-25 Data Communications Protocol Manual](https://www.microstrain.com/sites/default/files/3dm-gx5-25_dcp_manual_8500-0065_0.pdf)

I believe the device-supported command that will provide the functionality you're looking for is the GPS Time Update command (0x01, 0x72). Unfortunately it looks like support for this command has not yet been added to MSCL. I will add this to the backlog to include in the next release. Sorry for the inconvenience!

vibhoraggarwal, 2020-02-19:
> I believe the device-supported command that will provide the functionality you're looking for is the GPS Time Update command (0x01, 0x72)

Just for clarification, the command you mentioned is available as here: [getGPSTimeUpdate](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html#File:MicroStrain/Inertial/InertialNode.h:InertialNode.getGPSTimeUpdate) 

The functionality that we are missing in MSCL API is for the command: `GPS Correlation Timestamp (0x80, 0x12)`

Correct me if I am wrong, or you meant something else.

mglord, 2020-02-19:
Oh wow, sorry about that! I don't know how I missed that!

Yes, the InertialNode getGPSTimeUpdate and setGPSTimeUpdate should provide the functionality you're looking for (GPS Time Update command (0x01, 0x72)). To send the time update info to the GX5-25 you'll just need to replace:
`mscl::TimeUpdate timeupdate = node.getGNSSAssistTimeUpdate()`
with
`node.setGPSTimeUpdate(mscl::MipTypes::TIME_FRAME_SECONDS, <time of week in seconds>)`
The idea is that this would be sent once a second on the second to keep everything synchronized. The time cannot be updated with sub-second precision, so sending this more often could cause issues.
Upon initially connecting to the device it might be a good idea to send the command to synchronize the week number as well (use mscl::MipTypes::TIME_FRAME_WEEKS instead), but this is not needed in the main application loop.

The reference in the DCP to GPS Correlation Timestamp (0x80, 0x12) as a command is a mistake - I will pass this along to our documentation team for correction. The GPS Correlation Timestamp (0x80, 0x12) is a data field that can be output within the IMU data. This field as well as the Estimation Filter GPS Timestamp (0x82, 0x11) are both synchronized to the value sent via the setGPSTimeUpdate function.

Sorry for the confusion around this, hope this helps!

vibhoraggarwal, 2020-02-19:
> The reference in the DCP to GPS Correlation Timestamp (0x80, 0x12) as a command is a mistake

Thanks for the clarification. This was actually confusing me as well. I was trying to implement it using the [MSCL API](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20API%20Documentation/index.html), because there was not a direct mention of this in the API.  

> `mscl::TimeUpdate timeupdate = node.getGNSSAssistTimeUpdate()`
with
`node.setGPSTimeUpdate(mscl::MipTypes::TIME_FRAME_SECONDS, <time of week in seconds>)`

Currently I am not in the lab, I will check it tomorrow and update it here.

vibhoraggarwal, 2020-02-26:
> `mscl::TimeUpdate timeupdate = node.getGNSSAssistTimeUpdate()`
with
`node.setGPSTimeUpdate(mscl::MipTypes::TIME_FRAME_SECONDS, <time of week in seconds>)`

I implemented it in my code. Although the commands function and does not show an error, this commands does not help to send the timestamps from the GPS to the IMU; but, instead just stores and gets what information we are feeding as the GPStimestamps.

mglord, 2020-02-26:
My apologies, I'm afraid this may be beyond the scope of my specific product knowledge. I would encourage you to reach out to Barry Trutor ([Barry_Trutor@lord.com](Barry_Trutor@lord.com)) to provide better assistance with inertial product-specific questions. I'll also pass along this thread to him for context.
Barry should be able to help you figure out exactly what configuration and commands to use to achieve the desired result. Once that is determined, if you need further assistance with implementation in MSCL I would be happy to help! Sorry for the inconvenience!

vibhoraggarwal, 2020-03-02:
> `node.setGPSTimeUpdate(mscl::MipTypes::TIME_FRAME_SECONDS, <time of week in seconds>)`

This command was finally used for synchronizing the IMU time with that of the GPS.

Thanks @mglord and @barry_trutor for the support. We can close the issue if you don't have anything else to add.

mglord, 2020-03-03:
Thanks for updating here, glad we were able to help get this working!

closed 2020-03-03 by mglord