LORD-MicroStrain/MSCL #107: mscl::Connection::Serial Error: Connection error.

sandeepnpatil, 2020-02-20:
Hi,
I have build mscl for linux by following link https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Linux.md.
i compiled example code from following path "MSCL/MSCL_Examples/Wireless/C++/MSCL_Wireless_Example_C++/MSCL_Wireless_Example_C++"
I am able to compile properly but getting error in "mscl::Connection::Serial Error: Connection error"
I tried with 
1. 3000000 as baud
2. 921600 as baud
3. no baud
i am getting same error
[image]

sandeepnpatil, 2020-02-20:
after i run with sudo it is working.

mglord, 2020-02-20:
It sounds like your user may not have permission to access the serial port - one way to fix this is to add your user to the dialout group:
`sudo usermod -a -G dialout $USER`

The baud rate will depend on the base station you're connecting to. WSDA-200-USB communicates at a baud rate of 3000000 - if you're using a different base station the baud rate should be indicated on the datasheet under Connectivity.

Let me know if you continue to see the error after acquiring permissions to the port!

sandeepnpatil, 2020-02-21:
after running following command "sudo usermod -a -G dialout $USER"
still getting connection error.
but sudo ./main works.
not able to ping node.

mglord, 2020-02-21:
Sorry, I forgot to mention before that that permission change will not take effect until you log out and back in. If the issue continues would you mind adding a catch statement specifically for connection errors to print the error code as well? This may help determine what is happening:
```
catch (mscl::Error_Connection& e)
{
    cout << "Error: " << e.what() << " (" << e.code() << ")" << endl;
}
```
The error code should correspond to a Linux system error code.

**Re: unable to ping node**
Just to clarify - the issue is occurring when you run your application with sudo, right? Have you installed the WSDA linux [drivers](https://github.com/LORD-MicroStrain/Drivers/releases/tag/linux-cp210x-0.2)?

Would you be able to share your code where the 1) connection is established, 2) BaseStation and WirelessNode objects are initialized, and 3) node ping is attempted?

Also, can you successfully ping the BaseStation? What error is returned if you try a different command instead of ping, ie Set to Idle?

Vineet-Pandey, 2020-08-27:
Hi, I am facing a similar issue with the inertial sensor (GX-25-25 AHRS). Upon executing my code I get the following error:
```
 terminate called after throwing an instance of 'mscl::Error_Connection'
     what():  Connection error.
    Aborted (core dumped) 
```
Has this issue been resolved?

mglord, 2021-06-15:
@sandeepnpatil I'm going to go ahead and close this - I hope you were able to get it working! If you're still running into issues please feel free to re-open this and provide as much additional information as possible:

- Are you able to ping nodes when you run the application with `sudo`?
- Have you installed the WSDA linux [drivers](https://github.com/LORD-MicroStrain/Drivers/releases/tag/linux-cp210x-0.2)?
- Can you share your code for establishing the connection, instantiating the BaseStation and WirelessNode objects, and attempting to ping the node?
- Are you able to ping the Base Station successfully (`BaseStation::ping()`)?
- Do you get a successful response for any other command to the node ie Set to Idle or device info (`WirelessNode::setToIdle()`, `WirelessNode::serial()`)?

@Vineet-Pandey The most common connection issues are shared between wireless and inertial products, but troubleshooting often looks a little different. I hope you've been able to get this working, but if not would you mind opening another issue? it may be easier to discuss on a separate thread. Below are some general troubleshooting steps to try if you're still having trouble:

**Port Name**
The first thing to check is that the port name is correct. Port name formats:
- Windows: COM<number> for example COM8
- Linux: /dev/ttyACM<number> or /dev/ttyUSB<number> for example /dev/ttyACM0

**Confirm Permissions**
As mentioned above, this is a very common issue for Linux users - the port will not open if the user account does not have read/write permissions to it. There are various ways to check/setup permissions, but one that generally works is assigning the logged in user to the dialout group (Linux specific - see below).
`sudo usermod -a -G dialout $USER`
Note: the above permissions change will not take effect until the next time the user logs in

**Lookup System Error Code**
If the issue persists please try adding a try/catch around your Connection initialization and read out the associated error code. This code should correspond to a system error code and can be looked up for your operating system for additional info. Ex:
```
try
{
  mscl::Connection c(PORT_NAME, BAUD_RATE);
}
catch (mscl::Error_Connection& e)
{
    cout << "Error: " << e.what() << " (" << e.code() << ")" << endl;
}
```

closed 2021-06-15 by mglord