LORD-MicroStrain/MSCL #109: WSDA- Base-LXRS shuts down when measuring

maansuecia, 2020-03-13:
Hi,

I've a setup with basestation WSDA-Base-101 (Firmware 3.40) and a SG-Link-OEM-LXRS.
SO to the problem:
I've two WSDA-Base-101 (Firmware 3.40) that are behaving differently.
One of them seams to go to sleep after 10 - 30 minutes during the measurement. The other one never goes to sleep.

I've tested both basestation using "SensorConnect" and using a script, Python, and I'm getting the same behaviour.

Is there any setting to make sure that a basestation never goes to sleep during measurement ?

ardLORD, 2020-03-13:
> CAUTION: This email originated from a sender outside LORD Corporation. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hello,

Base stations should never go to sleep.  Are you using the analog outputs?  Are you seeing the data stop on the analog outputs, in SensorConnect or both?  Are you using the same node on both base stations?  If not, do you get the same behavior if you swap the nodes?

Thank you,
Please click here to rate the support you received!<https://www.surveymonkey.com/r/T2KPV6D>
Andrew Deering
Technical Support Engineer, Sensing Systems
+1(802)876-6546<tel:18028766546>
Andrew.Deering@Lord.com<mailto:ardeering@microstrain.com>

[Parker-LORD Logo]
www.microstrain.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.microstrain.com%2F&data=02%7C01%7CAndrew_Deering%40LORD.COM%7Cbff04fee00244dee1e4b08d7612db2aa%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637084721202086457&sdata=2rQxDwShCkTwzAMyFwIKxxcWncFI8A3VEgSBgx3d%2FnE%3D&reserved=0>
www.lord.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.lord.com%2F&data=02%7C01%7CAndrew_Deering%40LORD.COM%7Cbff04fee00244dee1e4b08d7612db2aa%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637084721202086457&sdata=dL72oR725C8fEHFNK8MRiNZK5cMXvn%2FHs2eYN5W%2Ba1s%3D&reserved=0>
www.parker.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.parker.com&data=02%7C01%7CAndrew_Deering%40LORD.COM%7Cbff04fee00244dee1e4b08d7612db2aa%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637084721202096450&sdata=X1to2tpWB36fD%2F83vIC58T7FI0MQgKZd4Gc%2Bi47TbVs%3D&reserved=0>
459 Hurricane Lane, Suite 102
Williston, VT 05495 USA

maansuecia, 2020-03-13:
Hi,
No, It's enabled but I'm not using the Analog Output. 
I just have a bridge simulator connected to the SG-LINK. The SG-LINK is the same in both tests.

ardLORD, 2020-03-13:
What are the serial numbers of these WSDA-101?

Download this older version of firmware( v3.37 ), downgrade the base station that has the issue and retest.

Thank you,
Please click here to rate the support you received!<https://www.surveymonkey.com/r/T2KPV6D>
Andrew Deering
Technical Support Engineer, Sensing Systems
+1(802)876-6546<tel:18028766546>
Andrew.Deering@Lord.com<mailto:ardeering@microstrain.com>

[Parker-LORD Logo]
www.microstrain.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.microstrain.com%2F&data=02%7C01%7CAndrew_Deering%40LORD.COM%7Cbff04fee00244dee1e4b08d7612db2aa%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637084721202086457&sdata=2rQxDwShCkTwzAMyFwIKxxcWncFI8A3VEgSBgx3d%2FnE%3D&reserved=0>
www.lord.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.lord.com%2F&data=02%7C01%7CAndrew_Deering%40LORD.COM%7Cbff04fee00244dee1e4b08d7612db2aa%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637084721202086457&sdata=dL72oR725C8fEHFNK8MRiNZK5cMXvn%2FHs2eYN5W%2Ba1s%3D&reserved=0>
www.parker.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.parker.com&data=02%7C01%7CAndrew_Deering%40LORD.COM%7Cbff04fee00244dee1e4b08d7612db2aa%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637084721202096450&sdata=X1to2tpWB36fD%2F83vIC58T7FI0MQgKZd4Gc%2Bi47TbVs%3D&reserved=0>
459 Hurricane Lane, Suite 102
Williston, VT 05495 USA

maansuecia, 2020-03-17:
Hi,
The Serial nr is 6307-1010-73834

ardLORD, 2020-03-17:
Did downgrading the firmware to v3.37 have any affect?

Thank you,
Please click here to rate the support you received!<https://www.surveymonkey.com/r/T2KPV6D>
Andrew Deering
Technical Support Engineer, Sensing Systems
+1(802)876-6546<tel:18028766546>
Andrew.Deering@Lord.com<mailto:ardeering@microstrain.com>

[Parker-LORD Logo]
www.microstrain.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.microstrain.com%2F&data=02%7C01%7CAndrew_Deering%40LORD.COM%7Cbff04fee00244dee1e4b08d7612db2aa%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637084721202086457&sdata=2rQxDwShCkTwzAMyFwIKxxcWncFI8A3VEgSBgx3d%2FnE%3D&reserved=0>
www.lord.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.lord.com%2F&data=02%7C01%7CAndrew_Deering%40LORD.COM%7Cbff04fee00244dee1e4b08d7612db2aa%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637084721202086457&sdata=dL72oR725C8fEHFNK8MRiNZK5cMXvn%2FHs2eYN5W%2Ba1s%3D&reserved=0>
www.parker.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.parker.com&data=02%7C01%7CAndrew_Deering%40LORD.COM%7Cbff04fee00244dee1e4b08d7612db2aa%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637084721202096450&sdata=X1to2tpWB36fD%2F83vIC58T7FI0MQgKZd4Gc%2Bi47TbVs%3D&reserved=0>
459 Hurricane Lane, Suite 102
Williston, VT 05495 USA

closed 2024-10-24 by github-actions[bot]