LORD-MicroStrain/MSCL #110: How to Reading sensor data on Arduino mega

Mohit953, 2020-03-22:
I want to read the sensor data using the Arduino mega. I am not getting how to start the implementation. Can anyone guide me with this?

Mohit953, 2020-03-23:
my aim is to integrate this to ardupilot. How to do that? My sensor is 3DM-CV5-25.

mglord, 2020-03-23:
I'm afraid MSCL and all it's dependencies will be too big to load onto the Arduino as part of your program, it was not designed for use on microcontrollers. I would recommend taking a look at our [MIP SDK](http://files.microstrain.com/MIP%20C%20Code%20Sample%20Windows%20and%20Linux.zip) - it does not specifically have examples for the CV5-25, but you may find the GX4-25 example project (MIP SDK\C\Examples\Windows\GX4-25\GX4-25_Test) useful, as well as the user guide (MIP SDK\C\Documentation\MIP SDK Users Guide.docx).

This should be used in combination with the CV5-25 [Data Communications Protocol document](https://www.microstrain.com/sites/default/files/3dm-cv5-25_dcp_manual_8500-0073.pdf) which describes the data fields and commands available for this device.

Regarding incorporating external code into your development environment and/or use within ArduPilot, your best bet is to find documentation specific to Arduino/ArduPilot development, but if you run into any problems related to our code we'd be happy to help resolve them. If you take a look at the above resources and need any clarification about anything let me know!

Mohit953, 2020-03-24:
If not arduino, STM32f1 discovery board, or Pixhawk board could do the work. 
Actually i am confused how to start this. Because, i donot want to try it with RPi board with ROS. I want to make it work with the pixhawk board or any STM32 family board.

Shubh1122000-coder, 2022-01-04:
Yes planning to read the imu yaw pitch and roll from this CV5-AR through STM32-F4 board as MSCL files are bit big and messy for me to understand so please provide the library for the same and pinout diagram for it.

closed 2024-10-24 by github-actions[bot]