LORD-MicroStrain/MSCL #111: Failed to read the LXRS+ Protocol Version (EEPROM 126) from Node 14911

RonNeal, 2020-03-25:
Failed to read the LXRS+ Protocol Version (EEPROM 126) from Node 14911 is an intermittent problem I am having when putting a wireless to sleep(). The firmware version in my wireless nodes is 12.44467, the Base Station's firmware version is 6.44079. MSCL version is 52.2.1 I have tried putting try - catch statements around the sleep() command but when this error occurs it still crashes the program. Any help on correcting this issue would be fantastic!

RonNeal, 2020-03-26:
I just now tried to duplicate the event I described above with SensorConnect and fortunately cannot. In my code the above fault ALWAYS happens when one of the nodes is sending a constant stream of data and another node is sporadically sending data.

ardLORD, 2020-03-27:
> CAUTION: This email originated from a sender outside LORD Corporation. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Hi Ron

Still waiting on the software team to look at your code.  I apologize for the delay and will forward their response as soon as it is available.

Thank you,
Please click here to rate the support you received!<https://www.surveymonkey.com/r/T2KPV6D>
Andrew Deering
Technical Support Engineer, Sensing Systems
+1(802)876-6546<tel:18028766546>
Andrew.Deering@Lord.com<mailto:ardeering@microstrain.com>

[Parker-LORD Logo]
www.microstrain.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.microstrain.com%2F&data=02%7C01%7CAndrew_Deering%40LORD.COM%7C54e5162bf4c5432edc3c08d7d15a5a53%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637208059943163455&sdata=DyUwv2tLYHSg7jIsXrEsImInrCLKX7EqOG84%2BN9pPPk%3D&reserved=0>
www.lord.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.lord.com%2F&data=02%7C01%7CAndrew_Deering%40LORD.COM%7C54e5162bf4c5432edc3c08d7d15a5a53%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637208059943173443&sdata=XH2BzBfjEKmxYOQx9PNWd1z%2FulQ8YvPp2X1S3V%2Bnjn4%3D&reserved=0>
www.parker.com<https://nam02.safelinks.protection.outlook.com/?url=http%3A%2F%2Fwww.parker.com&data=02%7C01%7CAndrew_Deering%40LORD.COM%7C54e5162bf4c5432edc3c08d7d15a5a53%7C239fbe1eac6d458cbe3c6d33d129b080%7C0%7C0%7C637208059943183440&sdata=4M0KmvwQNZKc0zYOrH3kMe7dypla%2FuaCbdyBEFRXktI%3D&reserved=0>
459 Hurricane Lane, Suite 102
Williston, VT 05495 USA

mglord, 2020-03-27:
Just for clarification regarding this description:

> the above fault ALWAYS happens when one of the nodes is sending a constant stream of data and another node is sporadically sending data

Are you trying to send the sleep command while the node is streaming?

A node that is asleep or streaming is only listening for the Set to Idle command - all other commands must be sent while the node is idle. If your node is streaming it must be set to idle prior to sending the Sleep command.

If this is not the problem, sharing your code around putting the node to sleep may help us narrow down what may be happening.

RonNeal, 2020-03-28:
No problem. Is there a place where all of the MSCL error codes are listed with what causes them?

mglord, 2020-03-30:
Assuming the issue you're seeing is still the Error_Communication exception "‘Failed to read the LXRS+ Protocol Version (EEPROM 126) from Node 14911’" there is limited information MSCL can provide due to the nature of the problem.

Communication errors with nodes means that MSCL sent the command to the base station but did not receive a response prior to timing out. This could mean any of the following, but in MSCL there is no way to tell the exact point of failure:
- Base station didn't send command to node
- Node didn't receive command
- Node didn't respond to command
- Base station didn't receive response
- Response was sent outside of MSCL timeout

The most common issue is the node not receiving the command - this could be caused by:
- Node is asleep
- Node is sampling
- Physical interference (distance, RF traffic)
- Low battery or inconsistent power supply
- Inherent inconsistencies in wireless communication

To ensure it's not a timeout issue or a one-off communication failure, you can try the following:

- Increase the node timeout (BaseStation::timeout()) - default: 20 ms
- Increase command auto-retries (WirelessNode::readWriteRetries()) - default: 0

RonNeal, 2020-03-31:
The cause is definitely when 2 nodes are sampling. If just the offending node is put to sleep with no other nodes sampling there is no issue. There seems to be no issue of setting multiple nodes to idle while sampling. I did correct the situation by adding a Boolean flag to the nodes class and after setting the node to idle the PutMeToSleep flag is set and the program proceeds as normal. A separate thread every few seconds reads through the array of active nodes and when it sees the PutMeToSleep flag it puts that node to sleep. This has worked flawlessly.

I had already increased the BaseStation timeout which did help but only a little. Is there an upper limit to this value?

Sorry to be so long winded.

Thanks!

Ron

mglord, 2020-03-31:
You should be able to put the node to sleep as soon as you receive the node's response from the Set to Idle command. Another way to tell if a node is currently idle is to ping it (WirelessNode::ping()) - this will not throw an exception on a communication failure, it will just indicate a failure in the return object (PingResponse::success()).

Sounds like you've found a solution that works for you though - that's great! If you'd like us to help determine the root cause anyway, it would be very helpful if you could provide your code around 1) initializing your network, 2) setting the node(s) to idle, and 3) putting the node(s) to sleep.

There's no upper limit to the base station timeout value, but at some point there's no reason a response would take so long. Andy (@ardLORD ) can probably be more helpful that I can in determining an appropriate timeout value for your configuration.

closed 2020-03-31 by RonNeal