LORD-MicroStrain/MSCL #115: Failed to read the Firmware Version (EEPROM 108) from the BaseStation

stephenagee, 2020-05-04:
When programming in VB (.NET 4.0 framework) the following exception is thrown when I attempt to set a node to idle. The exception is thrown when the node.setToIdle() function is called. There aren't any examples for VB programming in the SDK. Is the .NET 4.0 framework supported? The code is shown below:

`Public Class Form1
    Dim connection As mscl.Connection
    Dim baseStation As mscl.BaseStation
    Dim node As mscl.WirelessNode

    Private Sub Form1_Load(ByVal sender As System.Object, ByVal e As System.EventArgs) Handles MyBase.Load
        connection = mscl.Connection.Serial("COM34", 921600)
        baseStation = New mscl.BaseStation(connection)
        node = New mscl.WirelessNode(36939, baseStation)
    End Sub

    Private Sub btnStart_Click(ByVal sender As System.Object, ByVal e As System.EventArgs) Handles btnStart.Click
        System.Diagnostics.Debug.WriteLine("Setting node to idle")
        Dim idleStatus As mscl.SetToIdleStatus
        idleStatus = node.setToIdle() 'exception appears here
        While Not idleStatus.complete()
            System.Diagnostics.Debug.WriteLine(".")
        End While

        System.Diagnostics.Debug.WriteLine("Attempting to ping the Node...")
        Dim response As PingResponse = node.ping()
        If response.success() Then
            System.Diagnostics.Debug.WriteLine("Successfully pinged node!")
            System.Diagnostics.Debug.WriteLine("Base station RSSI: " + response.baseRssi())
            System.Diagnostics.Debug.WriteLine("Node RSSI: " + response.nodeRssi())
        Else
            System.Diagnostics.Debug.WriteLine("Failed to ping node")
        End If
    End Sub

End Class`

mglord, 2020-05-06:
**.NET version compatibility**
The pre-built MSCL .NET .dlls are built on .NET 3.5, but I have not run into issues before using it with newer versions - our example C# project targets .NET 4.5.2. It sounds like you've been able to build and run your application and successfully utilize some MSCL functions and objects so I suspect the issue is not due to a .NET compatibility problem.

**VB examples**
We do not have VB examples, but the C# example uses the same .NET library and can be used for reference when implementing MSCL functionality.

**Failed to read the Firmware Version (EEPROM 108) from the BaseStation**
Is this an intermittent problem or does the error occur every time?
What base station are you using? Can you try changing the specified baud rate from 921600 to 3000000?

stephenagee, 2020-05-06:
Changing the baud rate fixed it! For anyone reading this in the future, I'm using the WSDA-200-USB base station.

closed 2020-05-06 by stephenagee

vinod743374, 2021-12-07:
I am also facing the same issue ,
but the solution  doesn't work for me, it shows Connection error this time.