LORD-MicroStrain/MSCL #116: How to set slop and offset for a channel in vlink-200

sandeepnpatil, 2020-05-15:
i am using C++ mscl library, and referring below link of documentation.
http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20Documentation/index.html#File:MicroStrain/Wireless/Commands/AutoCalResult.h:AutoShuntCalResult.slope 

i need to set slop from my C++ application how can i do that?

mglord, 2020-05-19:
The function to set the slope and offset for wireless node channels is [WirelessNodeConfig::linearEquation()](http://lord-microstrain.github.io/MSCL/Documentation/MSCL%20Documentation/index.html#File:MicroStrain/Wireless/Configuration/WirelessNodeConfig.h:WirelessNodeConfig.linearEquation).

This requires two parameters: the channel mask representing the channel you'd like to set, and a LinearEquation object containing the slope and offset values.

The channel mask is a uint in which each bit represents a channel, for example:

- Channel 1 - decimal: 1, binary: 0000 0001
- Channel 3 - decimal: 4, binary: 0000 0100
- Channels 1, 4 - decimal: 9, binary: 0000 1001

To construct a channel mask you create an empty object then enable the channel(s) you want it to represent:
```
mscl::ChannelMask chMask;
chMask.enable(3); // enable channel 3
```

If you'd like to use the slope and offset determined by the auto shunt cal operation first use the WirelessNode::autoShuntCal() function with the same channel mask and target shunt cal parameters (ShuntCalCmdInfo) then build a LinearEquation object from the slope and offset within the AutoShuntCalResult. It will end up looking something like this (example for only channel 1):

```
// build channel mask
mscl::ChannelMask chMask;
chMask.enable(1);

mscl::ShuntCalCmdInfo shuntCalInfo;
// set desired parameters here

// run auto shunt cal operation on mscl::WirelessNode object (node)
mscl::AutoShuntCalResult result = node.autoShuntCal(chMask, shuntCalInfo);

// build WirelessNodeConfig object to set up configuration to apply
mscl::WirelessNodeConfig config;

// configure slope and offset for channel 1 based on shunt cal results
config.linearEquation(chMask, mscl::LinearEquation(result.slope(), result.offset()));

// apply configuration to node
node.applyConfig(config);
```

Hope this helps, let me know if anything is still unclear or you run into any further issues!

mglord, 2021-06-09:
Closing as resolved - if you continue to have trouble feel free to re-open this or create another issue. Thanks!

closed 2021-06-09 by mglord