LORD-MicroStrain/MSCL #117: 3DM-CX5-25 serial-TTL connection to PRUs of the PocketBeagle

EdgarBolivar, 2020-05-28:
Hi,

We would like to read a 3DM-CX5-25 using serial-TTL and the Programmable Real-Time Units (PRUs) in the PocketBeagle. To my understanding, these PRUs can be programmed in C but not C++. 

What would you recommend we can do to use the MSCL in the PRUs of the PocketBeagle?

Thank you!

mglord, 2020-06-22:
Sorry for the delayed response!

I'm not familiar with programming PocketBeagle PRUs, so there may be other options than those suggested here, but it sounds like the PRUs are separate microcontrollers within the PocketBeagle. A good rule of thumb is that anything without an operating system isn't going to be able to run MSCL - it needs to support Boost and you'll likely run into memory constraints.

I would recommend looking into using our [MIP SDK](http://files.microstrain.com/MIP%20C%20Code%20Sample%20Windows%20and%20Linux.zip) which is written in C - it does not specifically have examples for the CX5-25, but you may find the GX4-25 example project (MIP SDK\C\Examples\Windows\GX4-25\GX4-25_Test) useful, as well as the user guide (MIP SDK\C\Documentation\MIP SDK Users Guide.docx).

This should be used in combination with the [CX5-25 Data Communications Protocol](https://s3.amazonaws.com/download.microstrain.com/Manuals/3DM-CX5-25+DCP+Manual+(8500-0085).pdf) document which describes the data fields and commands available for this device.

I hope this helps! Let us know if you run into any issues!

EdgarBolivar, 2020-06-29:
This is great, thank you for the resources! 

Yes, the PRUs seem to be separate microcontrollers within the PocketBeagle with special access to some registers and peripherals. Your SDK is a perfect start for this kind of project.

Thank you again.

closed 2020-06-29 by EdgarBolivar