LORD-MicroStrain/MSCL #120: WSDA-200-USB working with Raspberry Pi

lushengsam, 2020-06-16:
I'm trying to get WSDA-200-USB to work with a Raspberry Pi 3 B, but without success. 

Here's what I have tried so far:
1). install the  Pre-built MSCL Binaries (v55.0.1) for Raspbian on the Raspberry Pi 3, which seems to be installed and working fine. 
2). Download and install the Linux driver from https://github.com/LORD-MicroStrain/Drivers/releases/tag/linux-cp210x-0.2 . The installation seems fine too, and after the installation, the Pi recognizes the WSDA-200-USB.
3). run the example code (SynchronizedSampling.py) in the MSCL repo.   
It failed at this line: node = mscl.WirelessNode(NODE_ADDRESS, baseStation)"  and says that it failed to read the Firmware Version from the EEPROM of the basestation.
I also tried a few other functions: it failed to ping the basestation; when trying to get the firmware version of the basestation using the firmwareVersion() function, it also says it failed to read the Firmware Version from the EEPROM. 

Is there a special driver for the Raspberry PI?  Is there something wrong with my configuration?

As information, my raspbian version is 10 (buster). 

Thank you.

mglord, 2020-06-22:
The problem is most likely caused by initializing the connection with the incorrect baud rate - the default is 921600 but the WSDA-200-USB communicates at 3000000. This is definitely something we will update to document more clearly.

Can you try adding the baud rate as a parameter when creating the Connection object?
`connection = mscl.Connection.Serial(COM_PORT, 3000000)`

Let us know if this doesn't resolve the issue and we'll go from there!

lushengsam, 2020-06-25:
Yes, that fixed it.  Thank you.

closed 2020-06-25 by lushengsam