LORD-MicroStrain/MSCL #121: Nodes become unresponsive after sending a few commands

Testing1User, 2020-07-10:
Hi,

Been having problems with my set-up recently, wondering if anyone can point me to the right direction. 
Using a WSDA-200-USB + V-Link-200 Setup. 
Latest MSCL 55.0.1 with the pre-built armhf (Raspbian) Debian Python 2 Package.

I can ping/sample/sleep my nodes no problem. After sending more commands the communication to the nodes drop and they become un-responsive. 

I think the issue is that the WSDA-200-USB LED stays on blue (sampling mode) even after I set all the nodes to idle. Is there a command to set it to idle so it goes back to green? Only way to go about it is to re-boot the system. 

Please let me know what system info + code you need and I will be happy to post them!
Thanks.

mglord, 2020-09-01:
Hi - so sorry for the delayed response!

The blue LED on the WSDA indicates that it is broadcasting a sync sampling beacon. This beacon can be turned off with BaseStation.disableBeacon(). Having the beacon enabled, however, should not interfere with node communication.

Do the nodes become unresponsive intermittently/seemingly at random, or is it always in a specific workflow or after a certain command or sequence of commands? Keep in mind that when a node is in sampling or sleep modes it will only respond to a set to idle command. The node configuration determines how often the node listens for the set to idle command in either of these modes (WirelessNodeConfig.checkRadioInterval()) and the WSDA needs to continuously send the set to idle command until the node responds to ensure it is heard (example [here](https://github.com/LORD-MicroStrain/MSCL/blob/master/MSCL_Examples/Wireless/Python/StopNode.py)).

If it's not an issue of the node being asleep/sampling, it would be helpful to know any additional details you can provide about the circumstances of the communication drop-outs.

Testing1User, 2020-10-23:
Hi,
 
I think the issue is -  cancelling my data parsing script and leaving it in sampling mode. 
Not calling the set to idle idle asap makes it so that I can't set the nodes to idle any more as it hits me with an EEPROM 108 error.

Cheers!

Testing1User, 2020-11-06:
Hi,
 
I get the EEPROM 108 error every now and then, which results with not being able to communicate with my nodes at all.
Sometimes, the script needs to be called a few times for it to work. However, there are times when no matter how many times I call a script it results in EEPROM 108 error.

Fastest way to re-create this error: 
1) Running & aborting the SetToIdle.py script in the terminal multiple times. Aborting the script with 'CTRL+C' before the setToIdle.complete() function returns true.

Fix:
1) Re-seat the WSDA.
2) Reboot the RPI.
3) Using the 'pkill -9' command to kill the script instead of 'CTRL + C' allows me to continuous call SetToIdle without any EEPROM 108 errors. 

I will close this issue as I have found a way around it. 
I only get the EEPROM 108 error without any success, calling a SetToIdle script after I have called the DataParse script via the Node-Red Exec function. I am currently using the hub-ctrl.c library to turn the ports off/on, which allows me to continue calling scripts after I get stuck in an error loop.

Cheers!

closed 2020-11-06 by Testing1User