LORD-MicroStrain/MSCL #122: Time Synchronization with External GPS Clock

ruiyishen2000, 2020-07-11:
I am using 3DM-GX5-10 for measurements in x, y, and z-accelerations. I am trying to implement it to print out a timestamp following each of the data packets.
Because the 3DM-GX5-10 does not come with timestamp functions, I am implementing this with an external GPS clock by synchronizing the external 1 PPS signal with the internal clock of the IMU and then try to print the timestamp associated with each data packet.
It was a little confusing when I was looking at the MSCL library that I am not exactly sure which functions should I be looking at. What would be the best way to implement this function that I want? I am using python3 for my codes.

mglord, 2020-09-01:
Hi - sorry for the delayed response!

InertialNode.setGPSTimeUpdate() should provide the functionality you need. For reference, this function calls the GPS Time Update (0x01,0x72) command - you can find more information about it in the [protocol documentation](https://www.microstrain.com/sites/default/files/3dm-cv5-10_dcp_manual_8500-0070.pdf). We recommend that the seconds update be sent once per second.

To output the timestamp, add the CH_FIELD_SENSOR_GPS_CORRELATION_TIMESTAMP channel field to the active channel fields (InertialNode.setActiveChannelFields()) at the same sample rate as the rest of your data. With this field included, it will be output as three of its own channels (CH_TIME_OF_WEEK, CH_WEEK_NUMBER, CH_FLAGS) as well as be assigned to the overall packet and can be read back with MipDataPacket.deviceTime().

I hope this helps! Let us know if you have any further questions regarding device behavior/setup or implementation!

ruiyishen2000, 2020-09-02:
Hi,

Thanks for your reply. I have been researching myself and tried using InertialNode.setGPSTimeUpdate() to implement the function that I am looking for. Upon reviewing the documentation, I am not sure how to put in the parameters for this function. I already have the GPS clock connected to the pin 7 and 8 of the IMU. 

-- offset | The <TimeFrame> object specifying what current time value to set, either weeks or seconds.
-- time | the current time.

How should these 2 parameters be related to the 1PPS signal that I got from the GPS clock?

Thanks again!

closed 2024-10-24 by github-actions[bot]