LORD-MicroStrain/MSCL #128: Manually setting Baudrate of Sensor.

Mopele, 2020-07-31:
Hey, I would like to set the baudrate for my 3DM-CV5-10 at every startup of an python script. How can I do that? 
Thanks for the help.

Mopele, 2020-08-03:
Apparently the only way to change the baurate is currently to set it via the MIP Monitor software form microstrain. It would be nice to also be able to set the baurate via Python to have for example a loop that checks for every bautrate and then resets the baurate to the desired one to get reliable 1kHz datarates. Also there is an firmware issue with the 3DM-CV5-10 where it doest store the set baudrate after a powerloss.

closed 2020-08-03 by Mopele

mglord, 2020-11-11:
Sorry for the delayed response - I know this is closed but I wanted to offer some clarification.

While the Connection object must be established with the device's currently configured baud rate, once the InertialNode object is created the function `InertialNode.setUARTBaudRate()` will change both the baud rate configured on the device and the connection baud rate (unless optional resetConnection parameter set to false).

All settings on our inertial devices are set only in temporary memory unless specifically saved as a startup setting. Any device reset or power loss will cause the device to boot back up in the startup configuration.

To save entire current configuration to startup: `InertialNode.saveSettingsAsStartup()`

To save only specific command/setting(s) to startup:
```
node = mscl.InertialNode(connection)

settings = mscl.MipTypes.MipCommands
settings.append(mscl.MipTypes.CMD_UART_BAUD_RATE)
# add any additional configuration command ids here

node.saveSettingsAsStartup(settings)
```

This save as startup functionality is also available in both SensorConnect (Save/Load Settings -> Save as Startup Settings) and MIP Monitor (Settings -> Save Current Settings...). Currently, both of these applications only support saving the complete current configuration to startup.

Sorry for the confusion - I hope this helps!