LORD-MicroStrain/MSCL #129: MSCL Lib integration Error

PrasathArunachalam, 2020-08-03:
We are developing system software application using Qt IDE, so we need to add MSCL C++ Lib in our application to log accelerometer data. Downloaded MSCL Lib from following link - **"http://github.com/LORD-Microstrain/MSCL", Version is pre-build MSCL binaries/packages(v55.0.1) windows: C++.

Added MSCL library. When we compiled we got following error and fixed some.

Error - 1 :- Winsock.h has been already included - FIXED
Error - 2 :- cannot open include file: boost/detail/endian.hpp: No such file or directory - FIXED by changing file path (Lib version - boost_1_73_0).
Error - 3 :- cannot open include file: openssl/config.h: No such a file or directory - FIXED by adding openssl lib (Lib version -> openssl-1.0.2j-fips-x86_64).
Error - 4 :- LINK2038: mismatch detected for 'Runtimelibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in main.obj. File not found: MSCL.lib (Version.obj) - NEED TO BE FIX.
Error - 5 :- LINK2038: mismatch detected for 'Runtimelibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in main.obj. File not found: MSCL.lib (version.obj) - NEED TO BE FIX.
Error - 6 :- LINK1104: cannot open file 'libboost_data_time-vc140-mt-x64-1_73.lib' - NEED TO BE FIX.

So i need help to fix above compile time errors. Also attached slides with errors screenshots for reference.

Thanks for the help team.

[MSCL Lib Integration ERR_Rev00_29July2020.pptx](https://github.com/LORD-MicroStrain/MSCL/files/5016999/MSCL.Lib.Integration.ERR_Rev00_29July2020.pptx)

mglord, 2020-08-05:
> Error - 4: LINK2038: mismatch detected for ‘Runtimelibrary’: value ‘MT_staticRelease’ doesn’t match value ‘MD_DynamicRelease’ in main.obj. File not found: MSCL.lib (Version.obj).
> Error - 5: LINK2038: mismatch detected for ‘Runtimelibrary’: value ‘MT_StaticRelease’ doesn’t match value ‘MD_DynamicRelease’ in main.obj. File not found: MSCL.lib (Version.obj).

This error is probably caused by a mismatch between MSCL and your project’s Runtime Library setting – MSCL is a static library so will only work if your project’s Runtime Library is set to Multi-threaded (/MT) or Multi-threaded Debug (/MTd) depending on your build configuration (Release, Debug). It looks like your current setting is Multi-threaded DLL (/MD). In Visual Studio this can be changed in project Properties under C/C++ -> Code Generation -> Runtime Library.

> Error - 6: LINK1104: cannot open file ‘libboost_date_time-vc140-mt-x64-1_73.lib’

MSCL depends on boost 1.68 – is there a specific reason you are hoping to use 1.73? You may be able to get it to work, but we cannot guarantee compatibility. Can you confirm that the libboost_date_time-vc140-mt-x64-1_73.lib’ exists in the directory you installed boost? The path should be something like this: <install directory>\boost_1_73_0\lib64-msvc-14.0\libboost_date_time-vc140-mt-x64-1_73.lib

It might be helpful to double check that you have completed all the steps in the Windows -> C++ section of our [How to Use MSCL](https://github.com/LORD-MicroStrain/MSCL/blob/master/HowToUseMSCL.md#c) page.

Let us know if it’s still unable to find the correct boot file(s) after confirming the file exists and the project configuration requirements are met!

karljohanreite, 2022-05-13:
Please see https://github.com/sintef-ocean/conan-mscl/commit/3f96a24181f86ea4dd37150900119ef0c66a13f0

closed 2024-10-24 by github-actions[bot]