LORD-MicroStrain/MSCL #131: (Question) How to get data remotely from WSDA-2000 using WebSocket ?

minida28, 2020-08-19:
I read in [this page]( https://github.com/LORD-MicroStrain/MSCL/blob/master/BuildScripts/buildReadme_Linux.md) that one can create secure WebSocket connections to an ethernet connected WSDA from anywhere in the world.

I have a WSDA-2000 connected to a 4G router via ethernet, the IP address is dynamic and the Sim card provider do now allow me to connect to it remotely (all ports are blocked I guess).
Right now I have figured out the way to access the WSDA-2000 control panel remotely (configured in sensorloud.microstrain.com).  How can I use MSCL to get the data remotely?

I did see there's a method to create connection using websocket using this method:

`connection = mscl.Connection.WebSocket(server, port)`

What server address to put there and what port? I cannot find any explanation about this, neither is WSDA manual, or in MSCL documentation.

Note that I have to manually compile the MSCL my self (I am using Ubuntu 18.04, python 2.7), because it seems the above method is not available in the provided pre-compiled MSCL binaries.

Thanks for the help in advance.

minida28, 2020-10-03:
Anyone?

mglord, 2020-11-13:
Sorry for the delayed response!

Unfortunately we're unable to support establishing a WebSocket connection via MSCL if the ports are blocked. 

It sounds like you were able to connect to the WSDA through SensorCloud - data that has been uploaded to SensorCloud can be pulled into your application for additional processing and analysis with the [SensorCloud API](https://github.com/LORD-MicroStrain/SensorCloud).

Hope this helps!

closed 2024-10-24 by github-actions[bot]