LORD-MicroStrain/MSCL #134: [Question]How to find the nodes on base station.

xidodongchen, 2020-10-06:
I connected the base station, but I can't know the node address connected there.
I want to find the node address through mscl library.

Testing1User, 2020-10-23:
Try using the code in Issue #77 
I've set one there before a while ago on discovering nodes. 
If I am not mistaken- the nodes send out data packets when they are first switched on. So run that script and try turning your node on and it should give you the address.

Rob-Hamilton, 2021-04-27:
So the only way for a Base Station to detect the nodes that are on the same frequency as it, is to go to every single node and turn them off and back on again? Seems like you should be able to broadcast a request to all nodes on the same frequency and get a response with the node information.

mglord, 2021-04-27:
I'm not a wireless communication expert, but my understanding is that a global ping is easier said than done due to interference. When sampling data from multiple wireless nodes they are carefully scheduled and synchronized with the base station's beacon to avoid sending data at the same time - I suspect there would be a similar issue here where multiple nodes responding at once would prevent the base station from picking up a complete response from any of them. I'll bring it up to our wireless team though in case I've misunderstood!

If you know the node addresses you can store a list of them and attempt to ping each one to detect if it's on-line (keeping in mind if the node is asleep or sampling 'set to idle' would need to be used instead to get a response).

Additionally, if the base station receives any data from a node, including diagnostic, the node address can be read from the packet and used to instantiate a new WirelessNode object. Obviously this is not an on-demand option but if nodes are in range, on the same frequency, and configured to transmit diagnostic data they will all be detected at some point. Unless disabled, diagnostic packets will be sent as long as the node is powered on, regardless of state (sleep, idle, sampling).

**Build WirelessNode from collected data:**
```
// this will run indefinitely - could be put in a separate thread of otherwise checked periodically instead
while (true)
{
    // get all data collected from the base station
    // baseStation is an instance of mscl::BaseStation
    mscl::DataSweeps data = baseStation.getData();

    // loop through data sweeps and check node addresses
    for (mscl::DataSweep sweep : data)
    {
        // could add logic to keep track of and check against previously discovered nodes
        std::cout << "found Node " << sweep.nodeAddress() << std::endl;

        // create WirelessNode instance with discovered address
        mscl::WirelessNode node(sweep.nodeAddress(), baseStation);
    }
}
```

Hope this helps!

Rob-Hamilton, 2021-04-28:
Thank you for the speedy response, it is appreciated. I am pretty new to GIT Hub and the LORD Sensors, still feeling my way through all of the documentation.

I was just thinking there is a way for the Base Station to detect nodes on other frequencies, it seems it should be a trivial matter to have it do the same but for it's own frequency. Just my thought. I haven't found out how it does that yet either though. Thanks again.

[image]

Rob-Hamilton, 2021-04-28:
I just experimented with that functionality. It would appear that it actually only knows the other nodes because it has at some time detected them on a Base Station and recorded the node information. It is not really going out to find those nodes either.

mglord, 2021-04-28:
Yes, that's accurate. The nodes on other frequencies are found through the same Node Discovery packets on power-on - the node sends a packet over every frequency and then SensorConnect remembers any that were received by a connected base station.

If you're running into any issues getting these strategies to work for your application and can let us know some of the specific challenges you're facing definitely reach out either here or to [product support](https://www.microstrain.com/contact-support) and we'd be happy to help figure out a solution!

closed 2024-10-24 by github-actions[bot]