LORD-MicroStrain/MSCL #136: [Question] Data logging high frequency in Raspberry Pi

Testing1User, 2020-10-23:
Hi, 

I am trying to log data from a G-Link-200 at 512Hz (all 3 channels). 
I have a WSDA-200-USB basestation plugged in a raspberry pi 3B running the latest mscl 58.0.5 python3.

Using a modified ParseData.py example script, I am inserting data sweeps into a CSV file (Using panda python library). I am only getting about ~1500-2000 ish points saved from each channel in about 60seconds which is less than the expected ~30k points. 
Is there a better method to go about logging datapoints on the pi?

Alternatively, I tried using the DatalogDownloader to download the sweeps- which took over 15minutes to download sampled data worth 30seconds stored in the G-Link!  

I suspect it could be the python execution speed or a hardware throttle.
If anyone could suggest me ways that could improve logging speed, I would greatly appreciate it!

mglord, 2020-11-11:
Hi - sorry for the delayed response!

**Re: saving to CSV**
If you set the node to idle after the 60 s and continue writing the buffered data (BaseStation.getData()) until there is none left do you get the expected (30k) number of data points? Just trying to determine whether something is going wrong in data transmission or if the actual logging is causing delays (log rate < collection rate).

**Re: DatalogDownloader**
This does seem excessive, but I'm not very familiar with development on the Raspberry Pi. Is the data being written to CSV during download? If so, does performance improve if you just store the points in a buffer without any additional processing?

If you want to share your code we can take a look to see if anything stands out. If you're not comfortable attaching your code here, feel free to email it ([melissa.gill@parker.com](melissa.gill@parker.com)).

If there are specific MSCL functions that seem to be heavily contributing to these performance issues, definitely let us know and we'll look into what we can do about it! Likewise, if you're experiencing unexpected behavior from the node itself let us know and we can help you track that down!

Testing1User, 2020-12-09:
Hi @mglord ,

Thank you for your response.

**Saving to CSV:**
There was nothing wrong with the data transmission. Looks like it was just logging it very slow due to me inserting the datasweeps to a CSV file. Your suggestion to set it to idle after 60seconds proved this correct.
**DatalogDownloader**
This problem was due to my coding again. Without implementing my CSV save function for the data downloader I can download 1minutes worth of stored data under 20seconds. Done by calling .complete() and timing how long it takes to reach 100%.

I have worked around this issue and everything is working fine now. I am now sampling for a given time and logging all my dataPoints into an array, then saving it into a CSV file. 

However, I am now logging raw values instead of the calibrated values from the G-Link. Is there a way I can get the calibrated values only ? 

Cheers!

mglord, 2020-12-10:
I'm glad to hear you got it working!

In order to get calibrated values the node's data format must be set to float. Were you logging calibrated values before making changes or has the behavior changed since?

To read the node's current configuration:
`WirelessNode.getDataFormat()`

To set the data format:
```
mscl::WirelessNodeConfig config;
config.dataFormat(mscl::WirelessTypes::DataFormat::dataFormat_cal_float);
WirelessNode.applyConfig(config);
```

Just to clarify - with your new code are you transmitting the data during sampling or logging it and using the DatalogDownloader afterwards?

If the data format is float but you're still seeing this behavior it may be helpful if you could export and send your node configuration - that may help us recreate the issue. If you have access to a Windows pc the easiest way to do this is to use the Export Configuration functionality in [SensorConnect](https://www.microstrain.com/software/sensorconnect). It may also be helpful to know the node's firmware version (`WirelessNode.firmwareVersion()`).

Testing1User, 2020-12-11:
Hi,

Thank you for your reply.
Here are my findings.

**1.** In order to get calibrated values the node's data format must be set to float. Were you logging calibrated values before making changes or has the behavior changed since?
**A.** For now I am logging calibrated float values in sensorconnect. Downloading data log sessions gives me the calibrated values in sensorconnect. Downloading the data log via the MSCL library (python script) gives me raw bit values (my guess).

**2.** To read the node's current configuration:
WirelessNode.getDataFormat()
**B.** I get a value 2 in return. According to the MSCL public decumentation -> dataFormat_cal_float  2 - Calibrated, float data format.

**3.** Just to clarify - with your new code are you transmitting the data during sampling or logging it and using the DatalogDownloader afterwards?
**C.** logging it and using the DatalogDownloader afterwards. 

**4.** Please see files attached below for the configurations.
Node Configuration (Calibration)
[Calibration.zip](https://github.com/LORD-MicroStrain/MSCL/files/5678224/Calibration.zip)
Node Configuration (Clone
[Clone.zip](https://github.com/LORD-MicroStrain/MSCL/files/5678226/Clone.zip)

WirelessNode.firmwareVersion() -> 12.44467.0

Testing1User, 2020-12-11:
Hi,

I managed to get my cal coefficients via the ChannelCalMap function. I used issue #26 as a reference.
However, I could not manage to get the channelID keys for the respective calCoefficient values.

**Code**
```
while not dataLog.complete():
    try:
        data1 = dataLog.getNextData()
        if dataLog.startOfSession():
            cal = dataLog.calCoefficients()

            for key, value in cal.items():
                print('{0} key corresponds to Unit: {1}'.format(key, value.unit()))
                print("Slope: ", value.linearEquation().slope())
                print("Offset: ",value.linearEquation().offset())
```

**Code output**
[image]

So I had to resort to getting my calCoefficients & their respective channelId using the mscl.WirelessNode().features()& iterating through the channelGroups to get the slope & offset.

```
            data1 = dataLog.getNextData()
            if dataLog.startOfSession():
                chGroups = node.features().channelGroups()
                for group in chGroups:
                    groupSettings = group.settings()
                    for setting in groupSettings:
                        if setting == mscl.WirelessTypes.chSetting_linearEquation:
                            le = node.getLinearEquation(group.channels())
                            print ("Linear Equation for: ", group.name())
                            print ("Slope: ", le.slope())
                            print ("Offset: ", le.offset())
```
Code output 

[image]

I will mark this issue closed for now.
I would like to thank you and the LORD-Microstrain team for their great work!

closed 2020-12-11 by Testing1User

mglord, 2020-12-11:
Thanks for including so much detail about the issue and the workaround you found!

Based on a quick search it's still unclear to me why logged float data would be transmitted uncalibrated, but it's possible the G-Link always stores logged data in integer format - I'll do some testing and talk to someone who's more familiar with this specific behavior to confirm this is working as designed.

Regarding reading the cal coefficients from the current settings instead of from the datalog session info:
The only risk here is if the cal coefficients have changed since collecting data. The coefficients, unit, and sample rate are stored for each datalog session so that if they are changed between sampling sessions the data is still meaningful when read back.

I'll look into the issue you're running into accessing the channel ID key in the datalog session calCoefficients and post an update here when we figure out what may be going wrong.