LORD-MicroStrain/MSCL #137: Error_InvalidSerialPort: Invalid Com Port with python and mscl on a symlink with udev rule

tguilment, 2020-10-29:
Hi the MSCL team,

I am working on the Jetson TX2 of nvidia with GNU/Linux. I want to connect the MicroStrain 3DM-GX5-25 without specifying the communication port. When I look in the `/dev` folder, I can see that the MicroStrain communication port is identified as` ttyACM0`. Then, if I unplug it and plug it later, the tty can then become `ttyACM1` for example.

In order to get a static name, I created a new udev rules in `/etc/udev/rules.d.`
I created a new file named `80-microstrain.rules` when I instruct the system to create a symbolic link.
In my case:
`SUBSYSTEM=="tty", ATTRS{idVendor}=="0483", ACTION=="add", KERNEL=="ttyACM*", SYMLINK+="ttyMicrostrain"`

I reboot the computer.
Now, when I plug the device, everything works well and I get a new ttyMicrostrain linked the ttyACM* created when the MicroStrain is plugged.
I can see it with 
`ls /dev/ that`
lrwxrwxrwx   1 root root           7 Oct 29 13:56 ttyMicrostrain -> ttyACM0

I am using the following python code:
```
import sys
sys.path.append('/usr/share/python3-mscl/')
import mscl as mscl
mscl.Connection.Serial('/dev/ttyMicrostrain')
```
I get:
```
Traceback (most recent call last):

  File "<ipython-input-4-a89a7daa943a>", line 15, in <module>
    connection = mscl.Connection.Serial('/dev/ttyMicrostrain')

Error_InvalidSerialPort: Invalid Com Port.
```

But if I use 
```
import sys
sys.path.append('/usr/share/python3-mscl/')
import mscl as mscl
mscl.Connection.Serial('/dev/ttyACM0')
```
then I don't get any errors.

How can I get a static way to communicate with the device?

Testing1User, 2020-11-02:
Try this -  COM_PORT = os.path.realpath("dev/custom_name") Issue #88 
or try issue #99

tguilment, 2020-11-02:
Thank you very much @Testing1User 
I first tried to use directly  
`COM_PORT = os.path.realpath("dev/custom_name")`
from my source directory which gave me the following path:
`/home/nvidia/Documents/dev/custom_name`

Then I just did:
```
# Save the current directory
s_cwd = os.getcwd()

# go to the '/' directory
os.chdir('/')

# Find the linked path
COM_PORT = os.path.realpath("dev/custom_name")

# Go back to the working directory
os.chdir(s_cwd)

```
Then I can see that `COM_PORT = '/dev/ttyACM0'`
If I unplug and plug the device again, I get `COM_PORT= '/dev/ttyACM1'`

I was looking for this kind of solution and that's working great!

closed 2020-11-02 by tguilment