LORD-MicroStrain/MSCL #138: Altitude Aiding enabled?: Error: The command (3399) is not supported.

tguilment, 2020-10-29:
I want to use the c++-mscl library in order to get data from the 3DM-GX5-25.
I am working on the Jetson TX2 of nvidia with GNU/Linux.
I manually built all the c++ dependencies and everything seems to be so far so good.
I downloaded the MSCL c++ examples for intertial node that you can find [here](https://github.com/LORD-MicroStrain/MSCL/tree/master/MSCL_Examples/Inertial/C%2B%2B/MSCL_Inertial_Example_C%2B%2B/MSCL_Inertial_Example_C%2B%2B)

The main.cpp is the following:
```
#include <iostream>
using namespace std;

#include "mscl/mscl.h"

#include "getCurrentConfig.h"
#include "parseData.h"
#include "setCurrentConfig.h"
#include "startSampling.h"
#include "setToIdle.h"

int main(int argc, char **argv)
{
    //TODO: change these constants to match your setup
    const string COM_PORT = "/dev/ttyACM0";

    try
    {
        //create a SerialConnection with the COM port
        mscl::Connection connection = mscl::Connection::Serial(COM_PORT);

        //create an InertialNode with the connection
        mscl::InertialNode node(connection);

        cout << "Node Information: " << endl;
        cout << "Model Name: " << node.modelName() << endl;
        cout << "Model Number: " << node.modelNumber() << endl;
        cout << "Serial: " << node.serialNumber() << endl;
        cout << "Firmware: " << node.firmwareVersion().str() << endl << endl;

        //TODO: Uncomment the lines below to run the examples

        //Example: Get Configuration
        getCurrentConfig(node);

        //Example: Set Configuration
        setCurrentConfig(node);       //Warning: this example changes settings on your Node!

        //Example: Start Sampling
        startSampling(node);

        //Example: Set to Idle
        setToIdle(node);

        //Example: Parse Data
        parseData(node);
    }
    catch(mscl::Error& e)
    {
        cout << "Error: " << e.what() << endl;
    }

    // system("pause");
    return 0;
```
I commented the `system("pause")` who doesn't not exist in linux.
Then, I compiled the code as follow:
I use the root privileges 
```
nvidia@nvidia:~$ sudo -i
[sudo] password for nvidia: 
```

I changed directory to be where the `main.cpp` is:
`root@nvidia:~# cd /home/nvidia/Documents/src/MSCL_Inertial_Example_C++`

I compiled the file
`root@nvidia:/home/nvidia/Documents/src/MSCL_Inertial_Example_C++# g++ -I/usr/share/c++-mscl/source -I/usr/share/c++-mscl/Boost/include main.cpp -o main -L/usr/share/c++-mscl -lmscl -lstdc++ -std=c++11`

Then I executed it and get the following:
```
root@nvidia:/home/nvidia/Documents/src/MSCL_Inertial_Example_C++# ./main
./main: error while loading shared libraries: libmscl.so: cannot open shared object file: No such file or directory
```
I fixed this error by adding the libmscl.so to `LD_LIBRARY_PATH` as follow:
```
root@nvidia:/home/nvidia/src# LD_LIBRARY_PATH=$LD_LIBRARY_PATH/usr/share/c++-mscl/
root@nvidia:/home/nvidia/Documents/src# export LD_LIBRARY_PATH
```

Then, as you will see, I executed the code 3 times in a row and I got 3 different results:
First time:
```
root@nvidia:/home/nvidia/Documents/src/MSCL_Inertial_Example_C++# ./main
Node Information: 
Model Name: Error: Failed to communicate with the device.
```

Second time:
```
root@nvidia:/home/nvidia/Documents/src/MSCL_Inertial_Example_C++# ./main
Node Information: 
Model Name: 3DM-GX5-25
Model Number: 6253-4220
Serial: 6253.67206
Firmware: 1.1.30

Error: Failed to communicate with the device.
```

Thrid time:
```
root@nvidia:/home/nvidia/Documents/src/MSCL_Inertial_Example_C++# ./main
Node Information: 
Model Name: 3DM-GX5-25
Model Number: 6253-4220
Serial: 6253.67206
Firmware: 1.1.30

AHRS/IMU Channels
-----------------

Estimation Filter Channels
--------------------------
Altitude Aiding enabled?: Error: The command (3399) is not supported.
```

I don't know/understand the error message.
Do you have an Ubuntu example that I can use?
Or, Is someone can guide me to make it work?

Besides that, I wish you the best during this special time.

mglord, 2020-11-16:
Sorry for the delayed response and confusion with our example code!

**Regarding the "Failed to communicate" errors**
It looks like you are able to communicate at least intermittently with the node as you are sometimes able to read the device information. The most common cause of this kind of behavior that I've seen is bandwidth issues if the device is trying to stream so much data that command response packets get dropped.

Is the device streaming (LED indicator flashing quickly) when you run the examples?

If this is the case, it may be helpful to add a set to idle loop in the beginning to establish more reliable communication, something like this:
```
while (true)
{
    try
    {
        node.setToIdle();
        break;
    }
    catch (mscl::Error_Communication) { /*ignore*/ } // may not get the response the first time, try again
}
```
Obviously in a real application implementation you would want to add some control logic to avoid looping forever in the event of an actual communication issue. In my experience there is no reason our inertial devices should need to call set to idle more than twice for a successful response (1st call successfully sets to idle, but response is dropped due to bandwidth issues, second call sets and responds successfully).

**Regarding "The command (3399) is not supported"**
Because each of our inertial devices support a different subset of MIP commands, not every function in the InertialNode class is supported by every device. It looks like our example code was made written based on the capabilities of the GX5-45. The following commands/functions in the examples are not supported by the GX5-25:

- getCurrentConfig.h
    - node.getAltitudeAid() *(hex: 0x0D47, decimal: **3399**)*
    - node.getAntennaOffset() *(hex: 0x0D13, decimal: 3347)*
- setCurrentConfig.h
    - node.setAltitudeAid(false)
    - node.setAntennaOffset(offset)

You can refer to the [GX5-25 Data Communication Protocol](https://www.microstrain.com/sites/default/files/3dm-gx5-25_dcp_manual_8500-0065_reference_document.pdf) document for additional information about all supported commands and data fields.

tguilment, 2020-11-17:
Thanks for your help! I could find the methods who work for me!
When I use the `setToIdle()` method I don't get the data. I get the data after I use the `resume()` method. After several tests and modifications, I finally got the following script working in my case! I can handle the potential instability by using some try/catch until the device is connected as you proposed. It also works if I stop the script a run it directly after. Moreover, with an udev rule I created (see how on the issue #137) I can even unplug then plugged the device and it will work so thanks!

```
#include <iostream>
using namespace std;

#include "mscl/Communication/Connection.h"
#include "mscl/Exceptions.h"
#include "mscl/MicroStrain/Inertial/InertialNode.h"

// Get the real path associated to the udev rule created for the MicroStrain.
#include <limits.h> /* PATH_MAX */

int main(int argc, char **argv) {

   char buf[PATH_MAX]; /* PATH_MAX incudes the \0 so +1 is not required */
    char *com_port = realpath("/dev/ttyMicrostrain", buf);
    if (com_port) {
        printf("This source is at %s.\n", buf);
    } else {
        perror("realpath");
        exit(EXIT_FAILURE);
    }

    // DONE: change these constants to match your setup
    const string port = com_port;
  mscl::uint32 baudrate = 921600;
  try {
    // create a SerialConnection with the COM port
    mscl::Connection connection = mscl::Connection::Serial(port, baudrate);

    // create an InertialNode with the connection
    mscl::InertialNode node(connection);

    bool success = node.ping();

    if (success) {
      std::cout << "Ping!" << std::endl;
    } else {
      std::cout << "[crickets]" << std::endl;
    }

    try
    {
        mscl::MipChannels estFilterChs; //ahrsImuChs;     estFilterChs.push_back(mscl::MipChannel(mscl::MipTypes::CH_FIELD_ESTFILTER_ESTIMATED_GYRO_BIAS, mscl::SampleRate::Hertz(100)));
        node.setActiveChannelFields(mscl::MipTypes::CLASS_ESTFILTER, estFilterChs);
        cout << "Channel activated" << endl;
    }
    catch (mscl::Error_Communication) {cout << "Channels already activated.." << endl;}

    while (true)
    {
	try
	{
	    node.resume();
	    break;
	}
	catch (mscl::Error_Communication) {cout << "Still not connected..." << endl;}
    }

    // endless loop of reading in data
    while (true) {
      // get all the data packets from the node, with a timeout of 500
      // milliseconds
      mscl::MipDataPackets packets = node.getDataPackets(500);

      for (mscl::MipDataPacket packet : packets) {
        // print out the data
        cout << "Packet Received: ";

        // get the data in the packet
        mscl::MipDataPoints data = packet.data();
        mscl::MipDataPoint dataPoint;

        // loop through all the data points in the packet
        for (unsigned int itr = 0; itr < data.size(); itr++) {
          dataPoint = data[itr];

          // print out the channel data
          cout << dataPoint.as_string()
               << " "; // Just printing this out as a string. Other methods (ie.
                       // as_float, as_uint16, as_Vector) are also available.

          // if the dataPoint is invalid
          if (!dataPoint.valid()) {
            cout << "[Invalid] ";
          }
        }
        cout << endl;
      }
    }
  } catch (mscl::Error &e) {
    cout << "Error: " << e.what() << endl;
  }

  // system("pause"); No pause command in linux
  return 0;
}
```

closed 2020-11-17 by tguilment