LORD-MicroStrain/MSCL #139: Writing in a specific address memory

gerson93, 2020-11-03:
Hi,

I'm trying to change the Region Code (EEPROM 280), in my bases. With SensorConnect is easy to achieve this, but now i'm want to develop a script in python to change the Region Code in my remote bases (running over ubuntu server 16.04). I'm trying to figure out how can i do this, but i can't find any help in the documentation.

Any ideas?
Thanks a lot,
BR.

gerson93, 2020-11-03:
well, it's very simple:

1. Once the connection is made with `baseStation = mscl.BaseStation(connection)`

2. then, we can use `baseStation.writeEeprom(address, value)` to write any value in a specific address, and `baseStation.readEeprom(address)` to read a value in a specific address.

I hope this is useful for someone
BR

mglord, 2020-11-16:
Thanks for posting the solution you used!

Most settings are wrapped for convenience in the `BaseStationConfig` and `WirelessNodeConfig` classes, but other specific EEPROM addresses can be directly read and written using the `read/writeEeprom()` functions.

EEPROM address definitions can be found [here](https://github.com/LORD-MicroStrain/Protocols) under Wireless -> [Wireless EEPROM Maps and Config Values](https://github.com/LORD-MicroStrain/Protocols/blob/master/Wireless/LORD%20MicroStrain%20Wireless%20Configuration.xlsx?raw=true).

Most users will not need to manually update EEPROM values - you should probably only do this if you are sure you know what you're doing or have been advised to by someone on our support team.

I'm going to go ahead and close this because it seems you've resolved the issue, but feel free to reopen or create additional issues if you have any additional questions!

Thanks!

closed 2020-11-16 by mglord