LORD-MicroStrain/MSCL #140: Is it possible to happen buffer overrun?

koniyasu, 2020-11-04:
I use 3DM-GX5-45(LORD Sensing Systems) and get data through MSCL.

When using this sensor, the program occasionally crashes. Sometimes it crashes after a long period of time, and sometimes it crashes after a short period of time.

There seems to be a buffer overrun in this library.

Also, I can't catch this error from outside the library.

Can you confirm this phenomenon and if it is a bug, can you please fix it?
Also, I would like to know if there is a way to catch the error.

Translated with www.DeepL.com/Translator (free version)

koniyasu, 2020-11-05:
Error information is here.
Camera lib is our library.

-----------------------
Unhandled Exception: System.AccessViolationException: Attempted to read or write
protected memory. This is often an indication that other memory is corrupt.
   at mscl.msclPINVOKE.Value_as_uint16(HandleRef jarg1)
   at mscl.Value.as_uint16()
   at CameraLib.PFunc`1.<.ctor>b__0_1(T g, MipDataPoint p)
   at CameraLib.PFunc`1.Set(T d, MipDataPoint p)
   at System.Linq.Enumerable.Count[TSource](IEnumerable`1 source, Func`2 predica
te)
   at CameraLib.GNSSSerialPort.ReadData()
   at CameraLib.GNSSSerialPort.WriteData(BinaryWriter& bw)
   at GrabCamera2.IMUDevice.run()
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionCo
ntext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, C
ontextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, C
ontextCallback callback, Object state)
   at System.Threading.ThreadHelper.ThreadStart()

mglord, 2020-11-16:
Thanks for getting the error info, and sorry for the delay!

I don't recall seeing this issue before, but will look into what may be causing it.

It may be helpful to know:
- What data channel fields you are streaming/processing
- The logic you are using when reading this data
    - ie Are you checking the `MipDataPoint.storedAs()` data type before accessing `as_uint16()`?
- Any additional information about how you are accessing the results of `InertialNode.getDataPackets()`
- It appears you're using Python - are you using one of our pre-built libraries or did you build it yourself? Are you using Windows or Linux?

closed 2024-10-24 by github-actions[bot]